calimero-project/calimero-core #23: Process communicator

AnaKatarina, 2015-07-01:
I have a weather station, which sends data to router and then router to Java, who must work continuously and follow all changes in weather, but there is a problem where communication between KNX router and Java application doesn't work when I click Terminate and run the app again. 
But when I click ENTER inside console, this line System.out.println("EXIT"); properly ends with this code: 
                             finally {
			    // we don't need the process communicator anymore, detach it from the link
			     if (pc != null)
			        pc.detach();
			   // close the KNX link
			     if (knxLink != null)
				knxLink.close();
		            }

So, my question is how to solve this problem? In the future user will not know how to properly end this app and it will run app again and it will not work, so do I need to put this code (finally) somewhere else? 
What is about this line: public void detached(DetachEvent e) {} 

I need something that will end application properly when user click Terminate, like I have "finally" part in code when it's clicked ENTER.

Please help me!

calimero-project, 2015-07-02:
The cause of this is the timeout mechanism for connections used in the KNXnet/IP protocol. The KNXnet/IP server will---for some time---keep your connection alive if you don't close it explicitly.

In your GUI, you will probably have some window listeners. For example, in Swing, using a `WindowAdapter`, it notifies you with the `windowClosing` method that the user is about to close the window. In such a method, just `close()` your `KNXNetworkLink`.

You don't need to call `detach` or act upon it. That method is basically for other communication logic to "attach" to, or "detach" from, a link. Because you close the application anyway, don't bother with that.

AnaKatarina, 2015-07-03:
Okey, thank you!

closed 2015-07-30 by calimero-project