calimero-project/calimero-core #25: Many unit tests are not working localy

ouaibsky, 2015-10-17:
Hi

I try to run tests  on climero core

> mvn clean install

And many are failing 
* Some of them because they found a knx gateway on my home network
* other reasons

It there any setup to apply before running the test ?
Thx

Christophe

bmalinowsky, 2015-10-27:
For most use-cases, its probably better to skip them, together with signing
```
mvn clean install -DskipTests -Dgpg.skip=true
```

Testing the communication protocols requires various endpoints at different layers. Most users won't have the complete required set of interfaces, e.g., for testing FT 1.2, or Routing, as well as endpoint setup. So, if an interface is not found, all (or almost all) unit tests for that category will fail.

You can see in some of the previous commits that I modified tests for a virtual subnet. This allows KNXnet/IP and KNX IP unit tests to execute completely in software. It requires the calimero-server and calimero-device. 
So yes, there exists a setup in software that allows most tests to run. But I cannot directly include the test setup, because then maven complains about circular dependencies (calimero-core <--> calimero-server <--> calimero-device).

Also note that a complete test run, due to tests of heart-beating, mgmt procedures, etc. takes around 20 minutes.

Another point you raised is with other KNX interfaces. Interfaces that do not participate in unit tests should actually be ignored. I will fix that in the test setup.

ouaibsky, 2015-10-29:
Ok, Thx for the update.

I can skip unit test, but it's nice to have it and safe when we want to do some refactoring.
Do you think it can be valuable to "group" UT (for example UT that require nothing special, those related to FT 1.2, ...) then to execute by default the default then execute other by activating some maven profile
Grouping can be done with junit category

My point is just to look if we can avoid to exclude all tests everytime.

For gpg it can make sens as well to disable it by default and activate only when u want to release. I'm gonna have a look.

teichsta, 2015-10-29:
> For gpg it can make sens as well to disable it by default and activate only when u want to release. I'm gonna have a look.

you could use maven profiles for that …

bmalinowsky, 2015-10-30:
UT grouping is probably a good way to go. 
Problem: IIRC(!) the only way to enforce grouping is via annotations. The current UTs are based on junit 3. And annotations on junit 3 tests are ignored. Mixing ju3/4 is not ideal. I will look into this.

For a quick workaround you might try to put exclusion patterns into a `property` of a new `profile` of your pom, together with the surefire plugin, which supports this:
http://maven.apache.org/surefire/maven-surefire-plugin/examples/inclusion-exclusion.html .
(This never worked reliably for me, but maybe it works for you.)

ouaibsky, 2015-10-30:
PR done for GPG: https://github.com/calimero-project/calimero.parent/pull/1

bmalinowsky, 2015-11-13:
gpg, junit categories, as well as unrelated knx interfaces during testing should be covered now (on j8 branch). Closing this.

closed 2015-11-13 by bmalinowsky