calimero-project/calimero-core #27: BCU/Tpuart + Calimero = IP Router?

tuxedo0801, 2015-11-04:
Hi there,

just wondering if it's somehow possible to attach an BCU/TpUart (like Siemens 5WG1 117-2AB11 
 --> https://www.siemens.be/cmc/upload/cms/docs/sbt/hvac/Brochures_Datasheets_manuals/FR/07_Building_Automation_Desigo/Datasheets/Desigo%20TRA%20-%20GAMMA_TPI_UP117_11_de.pdf) to rs232 /via usb to the PC and use it, together with calimero to get an "knx ip router"...?!

The mentioned BCU is already used in many DIY projects to access the knx bus. But if it's possible to combine it with calimero and get an ip router, this would be great.

The "eibd" from Martin Kögler  (https://www.auto.tuwien.ac.at/~mkoegler/index.php/eibd) supports this.

How about calimero?

br,
Alex

calimero-project, 2015-11-04:
v2.3 provides a link interface for TP-UART, see
https://github.com/calimero-project/calimero-core/blob/master/src/tuwien/auto/calimero/link/KNXNetworkLinkTpuart.java

tuxedo0801, 2015-11-04:
Great. So accessing KNX via TP-UART would be possible. 

I guess the existing IP Link implementation (tunneling+routing) can only interact with a real device and can not be used as a kind of "master" so that others can send data to this master, which can be then routed to the TP-UART link and vice versa?

calimero-project, 2015-11-04:
I don't fully understand what you consider a real device.

Depending on the KNXnet/IP protocol requested to be used by the IP link, 1 endpoint (Tunneling) or [0..*] endpoints (Routing) are required. Client- and server-side of the protocols differ.

From your first question about a knx router interface, for calimero/eibd this means: 
eibd uses the EIBnet/IP server front end, for which the equivalent in calimero is the KNXnet/IP server. 
If you want to use the eibd front end for marshalling client data over a socket, you have to use eibd as this is specific to eibd.
How you multiplex data for transmission over a tpuart link is up to the application.

tuxedo0801, 2015-11-04:
Seems we are using different wording. And I must admit, that I'm maybe not so familiar with the knx-conform-wording(, maybe also because I don't differentiate between knx and eib?!)

Question was: Is it possible to use an KNX TP-UART + Calimero + the computer's network-interface to "behave like an KNX IP Router device": Traffic from wired-knx-bus is routed to the ethernet (knx ip) and vice versa ??

From what I already understood:

The implementation to talk to the knx bus with an TP-UART is available. ok.

And the other end seems also to be present? --> "for which the equivalent in calimero is the KNXnet/IP server. "

Seems that all required part's are already there?
So, to put it all together:

Data from knx bus receives via TP-UART and the corresponding TP-UART link implementation in calimero. All I have to do is, forward the data (and maybe increment the routing counter and eventually some other stuff that real KNX IP Routers do) via the "KNXnet/IP server"(???) implementation to the ethernet and vice versa, right?

calimero-project, 2015-11-04:
> Is it possible to use an KNX TP-UART + Calimero + the computer's network-interface to "behave like an KNX IP Router device": Traffic from wired-knx-bus is routed to the ethernet (knx ip) and vice versa ??

Yes, take a look at the readme of [calimero-server](https://github.com/calimero-project/calimero-server). It does exactly what you want. Maybe a little ascii art helps:
```
                          |             calimero-server                |
                          |    server-side           KNX subnet side   |
client <-KNXnet/IP link-> | KNXnet/IP <--> gateway <--> [1..n] subnets | <-link-> KNX
```

> All I have to do is, forward the data (and maybe increment the routing counter and eventually some other stuff that real KNX IP Routers do) via the "KNXnet/IP server"(???) implementation to the ethernet and vice versa, right?

This is the case if you only run the server-side parts of the KNXnet/IP _protocol_ (as opposed to running a KNXnet/IP server). Then you have to implement all the rest that makes up an actual "server", i.e., the gateway logic, interface objects, subnet connection, message filtering, etc.

tuxedo0801, 2015-11-13:
Thanks for the details.

closed 2015-11-13 by tuxedo0801