calimero-project/calimero-core #30: Changing write for DPTXlator from private to public

benjaminaigner, 2015-12-02:
Is it possible and wise to change the visibility of following method to public?

ProcessCommunicatorImpl.write(final GroupAddress dst, final Priority p, final DPTXlator t)

If I use the implementation in an external project, it would like to have the full flexibility of the DPTXlator classes for writing to the KNX bus. By now this is not possible due to the private declaration of this method.

bmalinowsky, 2015-12-04:
You already have the full flexibility for translation by using `write(GroupAddress, DPTXlator)`, no? Basically you want to also set a different priority for a specific write. It's certainly possible, but I don't know if that's wise or not. AFAIK, a different priority is rarely used, but I could be wrong. So yeah, I'm more or less neutral wrt that. Currently reading/writing a _datapoint_, i.e., `read/write(Datapoint [, String])` is setting the priority. I actually would opt for that, because a priority is assigned to a datapoint, not a single message.

benjaminaigner, 2015-12-04:
I was using the latest release, it is fixed in the current master branch :-)

Sorry for that!

closed 2015-12-04 by benjaminaigner

bmalinowsky, 2015-12-04:
No prob :)