calimero-project/calimero-core #31: Awaiting 2.3 release ...

tuxedo0801, 2015-12-24:
In reference to issue #28 --> Is there any update on this? Will the release come this year?

tuxedo0801, 2016-01-26:
*ping* @bmalinowsky ...

tuxedo0801, 2016-02-09:
Is this project kind of dead?!

bmalinowsky, 2016-02-12:
> Is this project kind of dead?!

It's kind of not?!

tuxedo0801, 2016-02-12:
So, if it's not dead ;-) Any ideas about the 2.3 release date?!

bmalinowsky, 2016-02-12:
There will be a beta before, some days from now. Reason being that accounting for repeated .inds with TP-UART does affect execution time in some cases (might trigger timeouts and degrade performance). Release depends on a reasonable fix for that.

bmalinowsky, 2016-02-12:
... and when i replace a siemens relay that melted half the TP lab setup

tuxedo0801, 2016-02-15:
Thanks for the update, will look forward to the beta.

calimero-project, 2016-02-16:
v2.3-beta done. Please open a new issue for the beta where necessary.

closed 2016-02-16 by calimero-project