calimero-project/calimero-core #32: Connecting with multicast interface requests to the bus are always flaged with repeated=true

lewie, 2016-04-10:
Hello @bmalinowsky,

when connecting calimero by multicast interface to a KNX bus, all requests (respond or write) coming from calimero have set the "RepeatedFlag"  flag to true. Debugged with ETS5.

This is definitively not correct, because it is not a repeated!
It also has the side-effect, that repeated request without previous external bus transactions from other devices, can not be answered by calimero.

It seams with a connection by TUNNEL, requests are correctly flagged.

I connected with several calimero tools and examples (ProcComm and PushButtonActuator etc.) same effect. I only can prevent it hardcoded over CEMILData.setRepeat().

I do not understand all backgrounds to fix the error in calimero.

Thank you in advance.
Helmut

bmalinowsky, 2016-04-10:
Hi, thanks I will look into that.

Can you elaborate on the side-effect you mentioned wrt responses that cannot be answered by calimero? Especially the setup (KNX subnet medium) and calimero version you use? The reason is, you mention the PushButtonActuator example, it uses KNX IP. The repeat flag is not relevant here, as it does not apply on that medium.

For now, I tried the push-button example, but it updates and answers fine from what I can see.

lewie, 2016-04-11:
Hi,
using KNX IP as ROUTER as multicast interface (224.0.23.12:3671) then *RepeatedFlag=true* at all communication to the knx bus.

**Log ETS5 *RepeatedFlag=true*:**  Makes no sense.
[image]

**Log ETS5 *RepeatedFlag=true* and READ does not work without Busaction:**: The side-effect.
[image]

**Log ETS5 READ does work correct:**: Calimero 2.3 beta, hardcoded RepeatedFlag always false
[image]

To force calimero to RepeatedFlag=false I manipulated tuwien.auto.calimero.cemi.CEMILData::setRepeat().

Think now it is more clear what I mean. ;-)
Helmut

PS:
> For now, I tried the push-button example, but it updates and answers fine from what I can see.

Does push-button example answer a groupReadRequest coming from the bus? Mine doesn't:

````
/*
    Calimero 2 - A library for KNX network access
    Copyright (c) 2015 B. Malinowsky

    This program is free software; you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation; either version 2 of the License, or
    (at your option) any later version.

    This program is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program; if not, write to the Free Software
    Foundation, Inc., 59 Temple Place, Suite 330, Boston, MA  02111-1307  USA

    Linking this library statically or dynamically with other modules is
    making a combined work based on this library. Thus, the terms and
    conditions of the GNU General Public License cover the whole
    combination.

    As a special exception, the copyright holders of this library give you
    permission to link this library with independent modules to produce an
    executable, regardless of the license terms of these independent
    modules, and to copy and distribute the resulting executable under terms
    of your choice, provided that you also meet, for each linked independent
    module, the terms and conditions of the license of that module. An
    independent module is a module which is not derived from or based on
    this library. If you modify this library, you may extend this exception
    to your version of the library, but you are not obligated to do so. If
    you do not wish to do so, delete this exception statement from your
    version.
*/

import tuwien.auto.calimero.GroupAddress;
import tuwien.auto.calimero.IndividualAddress;
import tuwien.auto.calimero.exception.KNXException;
import tuwien.auto.calimero.datapoint.Datapoint;
import tuwien.auto.calimero.datapoint.StateDP;
import tuwien.auto.calimero.device.BaseKnxDevice;
import tuwien.auto.calimero.device.KnxDeviceServiceLogic;
import tuwien.auto.calimero.dptxlator.DPT;
import tuwien.auto.calimero.dptxlator.DPTXlator;
import tuwien.auto.calimero.dptxlator.DPTXlatorBoolean;
import tuwien.auto.calimero.dptxlator.TranslatorTypes;
[... 109 lines cut ...]
		catch (final KNXException e) {
			System.out.println("Initializing " + actuatorName + " failed: " + e.getMessage());
			e.printStackTrace();
		}
		finally {
			System.out.println(actuatorName + " has left the building.");
		}
	}

	@Override
	public void run()
	{
		// wait forever, until interrupted
		try {
			synchronized (this) {
				while (true)
					wait();
					//Thread.sleep(1000);
			}
		}
		catch (final InterruptedException e) {}
	}

}
````

bmalinowsky, 2016-04-11:
That routing indications show up as repeated makes no sense, thank you for pointing that out! Should be fixed with @97ea2e20e3c67f110c6219639280d97409d9fd04.

However, I tested two setups, 1) using a pure KNX IP network with your pushbutton example, as well as 2) a routing backbone attached to PL110. In both cases the repeat flag made no difference (if it's set or not, which makes sense).

bmalinowsky, 2016-04-11:
Wrt. EDIT2: you use the standard process communicator. For example with the pushbutton, switch it every 5 seconds in `run()` (make `device` a field):

```
while (true) {
	wait(5000);
	try (ProcessCommunicator pc = new ProcessCommunicatorImpl(device.getDeviceLink())) {
		pc.write(dpAddress, state ^= true);
	}
	catch (KNXLinkClosedException | KNXTimeoutException e) { e.printStackTrace(); }
}
```

lewie, 2016-04-11:
In my push-button Setup (ROUTER as multicast interface (224.0.23.12:3671)) it works perfect now!
Later I will test it with TUNNELING Setup and in OpenHAB Bundle.

Your example with ProcessCommunicator works perfect!

Thank you very much!
Helmut

lewie, 2016-04-11:
OK, TUNNELING is working too in OpenHAB it works perfect. We can close this.

I had forgotten:
When I use the original example PushButtonActuator.java I get:

> Multiple markers at this line 140
> 	- The resource type KNXNetworkLink does not implement 
> 	 java.lang.AutoCloseable
> 	- The resource type KNXNetworkLinkIP does not implement 
> 	 java.lang.AutoCloseable
Add  extends AutoCloseable to KNXNetworkLink helped me.

bmalinowsky, 2016-04-11:
Wrt EDIT2, part 2: ofc, there might be reasons to override dispatching and threading of service notifiers. If so, please open a new issue to allow that. Otherwise, it's basically an implementation detail. ProcessCommunicationResponder is used to send AL responses (and only those!).

PushButtonActuator:
> Multiple markers at this line 140
>
> The resource type KNXNetworkLink does not implement java.lang.AutoCloseable
> The resource type KNXNetworkLinkIP does not implement java.lang.AutoCloseable

Yes, it's a Java 8 example, so it uses Calimero Java SE Embedded 8, not the old Java ME CDC (where Closeables were not born yet ;).

closed 2016-04-11 by bmalinowsky