calimero-project/calimero-core #33: KNXTimeoutExceptions when connected by TUNNELING

lewie, 2016-04-12:
Hello @bmalinowsky,
dispatching a new state (using Puschbutton, as you described [here ](https://github.com/calimero-project/calimero-core/issues/32#issuecomment-208400461)) throws KNXTimeoutExceptions when connected by TUNNELING:

```
tuwien.auto.calimero.exception.KNXTimeoutException: no confirmation reply received for L-Data.req from 1.1.0 to 1/1/10, low priority hop count 6 repeat tpdu 00 81
	at tuwien.auto.calimero.knxnetip.ClientConnection.doExtraBlockingModes(ClientConnection.java:236)
	at tuwien.auto.calimero.knxnetip.ConnectionBase.send(ConnectionBase.java:269)
	at tuwien.auto.calimero.knxnetip.KNXnetIPTunnel.send(KNXnetIPTunnel.java:149)
	at tuwien.auto.calimero.link.KNXNetworkLinkIP.onSend(KNXNetworkLinkIP.java:263)
	at tuwien.auto.calimero.link.AbstractLink.send(AbstractLink.java:304)
	at tuwien.auto.calimero.link.KNXNetworkLinkIP.sendRequestWait(KNXNetworkLinkIP.java:240)
	at tuwien.auto.calimero.process.ProcessCommunicatorImpl.write(ProcessCommunicatorImpl.java:466)
	at tuwien.auto.calimero.process.ProcessCommunicatorImpl.write(ProcessCommunicatorImpl.java:262)
	at intro.PushButtonActuator2$1.run(PushButtonActuator2.java:327)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
```
By default ROUTER connections never confirm, there fore it is only an issue for TUNNELING.

Only can understand half the requirements from code and this [discussion](https://sourceforge.net/p/calimero/discussion/485103/thread/046a3896/).

Should I ignore the KNXTimeoutExceptions or is there a way to get the confirmations?

Thank you in advance.
Helmut

bmalinowsky, 2016-04-12:
You should not ignore confirmations, as in a correct setup the server should supply if everything is ok. `1.1.0` is your router device address, right? What if you initialize the link just with `new TPSettings()`. You get a .con then?

lewie, 2016-04-13:
AHHH, now I understand.
For connection type TUNNEL:
If 0.0.0 is set as device address, means no/unset Address.
Then the Tunnel addresses provided by the router will be used.

It works now.
TYVM

closed 2016-04-13 by lewie