calimero-project/calimero-core #40: dpt 229.001 isn't supported

Darkwa, 2017-02-13:
Hello,
I'm trying to work with an energy-meter. It uses the DPT 229.001
This dpt seems to be quite new (2014).
Informations about this DPT can be found in the PDF below, page 75 :
[pdf](http://www.sti.uniurb.it/romanell/Domotica_e_Edifici_Intelligenti/110504-Lez10a-KNX-Datapoint%20Types%20v1.5.00%20AS.pdf)

Is there a way to deal with this DPT for now?

Thanks,
Pierre

bmalinowsky, 2017-02-14:
I would extract `CountVal`, use the signed 32 bit translator (`DPTXlator4ByteSigned`), and hardcode the scaling depending on your metering device. For the `Status/Command`, I created a branch `feat/xlator-8bitset` with a basic implementation of the bitset translator; use `DptGeneralStatus`.

Darkwa, 2017-02-15:
Okay thanks, I'm gonna try that !

bmalinowsky, 2017-02-16:
The branch now also contains a first implementation for translator 229.x. Maybe you can give it a try and provide feedback on any issues.

Darkwa, 2017-03-08:
Hello !
I come back just to let you know that since it was the last day of my project, I have not been able to try this with the KNX network.
After my presentation, I totally forgot this thread, sorry.
Thanks again, I was suprised some months ago when I discovered that people were still working on Calimero !

calimero-project, 2017-03-09:
No problem, i hope your presentation went well :)

closed 2017-03-09 by calimero-project