calimero-project/calimero-core #41: Reading/writing value from/to a KNX module

JourdanM, 2017-02-23:
Hello !

I'm currently trying to read/write a value from/to a KNX module. I tried to follow the steps from this post (https://github.com/calimero-project/calimero-core/issues/21) but I've issues because the basic code source has changed (remoteHost isn't anymore a String but a InetAdress per example).

Is it possible to update these parts of the code so I could try to make my application work ? 

Thanks a lot and sorry for my bad english :D

calimero-project, 2017-02-23:
This is an updated example:

	import java.net.InetSocketAddress;
	
	import tuwien.auto.calimero.GroupAddress;
	import tuwien.auto.calimero.KNXException;
	import tuwien.auto.calimero.link.KNXNetworkLink;
	import tuwien.auto.calimero.link.KNXNetworkLinkIP;
	import tuwien.auto.calimero.link.medium.TPSettings;
	import tuwien.auto.calimero.process.ProcessCommunicator;
	import tuwien.auto.calimero.process.ProcessCommunicatorImpl;
	
	/**
	 * Example of Calimero process communication, we read and write a boolean datapoint in the KNX network. By default, this
	 * example will not change any datapoint value in the network.
	 */
	public class ProcessCommunication
	{
		// Address of your KNXnet/IP server. Replace the IP host or address as necessary.
		private static final String remoteHost = "myKnxServer.myHome.com";
	
		/**
		 * We will read a boolean from this KNX datapoint group address, replace the address string with one of yours. Make
		 * sure this datapoint exists, otherwise you will get a read timeout!
		 */
		private static final String group = "1/1/1";
	
		public static void main(final String[] args)
		{
			// Create our network link, and pass it to a process communicator
			try (KNXNetworkLink knxLink = KNXNetworkLinkIP.newTunnelingLink(null, new InetSocketAddress(remoteHost, 3671), false, TPSettings.TP1);
				ProcessCommunicator pc = new ProcessCommunicatorImpl(knxLink)) {
	
				System.out.println("read a boolean from datapoint " + group);
				// this is a blocking method to read a boolean from a KNX datapoint
				final boolean value = pc.readBool(new GroupAddress(group));
				System.out.println("datapoint " + group + " value: " + value);
	
				// Uncomment the next line, if you want to write back the same value to the KNX network
				// pc.write(group, value);
			}
			catch (KNXException | InterruptedException e) {
				System.out.println("Error reading KNX datapoint: " + e.getMessage());
			}
		}
	}

JourdanM, 2017-02-24:
Thanks a lot, it works fine, I'm now able to collect data from differents groups !
When I tried do run my app from the school (directly connected by Wi-Fi on the network), I have a timeout, impossible to connect to the group. 
But when I tried from home (connected by VPN), it works fine.

Do you have an idea why ?

bmalinowsky, 2017-02-24:
The example probably defaults to the wrong local IP address. 
In the method call `newTunnelingLink(null, ... `, replace the null with `new InetSocketAddress("my.local.ip.address", 0)`, and insert your local IP address of the wifi network.

JourdanM, 2017-02-24:
Works well :) 
Thank you very much for the help !

closed 2017-02-24 by JourdanM