calimero-project/calimero-core #42: Getting values from KNX module

JourdanM, 2017-03-07:
Hello 

I'm currently working with Calimero (which is pretty useful!) and I've a question about how to get value from a KNX module.

My Weather Station have sensors (temperature, humidity etc.) and I'm now able to ask a value using the librairy : 
client (Calimero) --------> KNX module (Weather Station)
client (Calimero)<-------- temperature (p. ex.)

I'd like now to get values in a different way, this one : 
client (Calimero)<-------- KNX module (Weather Station)
client (Calimero) --------> temperature (p. ex.)

Description : the Weather Station send to the client the temperature (p. ex.), when the values changes (is it possible, may I set in ETS that I'd like the value when it changes from 0.5°C p. ex. ?)

I also configure the KNX module to send on the bus the value every hour. Is it possible to be notify in Calimero ?

This is kind of a monitoring app. 

Thanks in advanced if you have some examples/informations about this problem.

Have a goo day

calimero-project, 2017-03-09:
Hello,
> Hello
>
> I'm currently working with Calimero (which is pretty useful!) and I've a question about how to get value from a KNX module.
>
> My Weather Station have sensors (temperature, humidity etc.) and [...]
You just have to add a process listener to your process communicator. It will listen to all changes on the knx network.

Based on the last example, we can extend it like this:

public class GroupMonitor implements ProcessListener
{
	/**
	 * Address of your KNXnet/IP server. Replace the host or IP address as necessary.
	 */
	private static final String remoteHost = "192.168.10.10";

	// Main entry-level routine
	public static void main(final String[] args)
	{
		// create an instance of the group monitor and run it
		new GroupMonitor().run();
	}

	public void run()
	{
		final InetSocketAddress remote = new InetSocketAddress(remoteHost, 3671);
		try (KNXNetworkLink knxLink = KNXNetworkLinkIP.newTunnelingLink(null, remote, false, TPSettings.TP1);
				ProcessCommunicator pc = new ProcessCommunicatorImpl(knxLink)) {
			System.out.println("Monitoring KNX network using KNXnet/IP server " + remoteHost + " ...");
			// start listening to group notifications using a process listener
			pc.addProcessListener(this);
			while (true)
				Thread.sleep(1000);
		}
		catch (final KNXException | InterruptedException | RuntimeException e) {
			System.err.println(e);
		}
	}

	public void groupWrite(final ProcessEvent e) { print("write.ind", e); }
	public void groupReadRequest(final ProcessEvent e) { print("read.req", e); }
	public void groupReadResponse(final ProcessEvent e) { print("read.res", e); }
	public void detached(final DetachEvent e) {}

	// This method is called on every group notification issued by a datapoint on the KNX network.
	// It prints the service primitive, KNX source and destination address, and Application Service
	// Data Unit (ASDU) to System.out.
	private void print(final String svc, final ProcessEvent e)
	{
		try {
			System.out.println(LocalTime.now() + " " + e.getSourceAddr() + "->" + e.getDestination() + " " + svc + ": "
					+ DataUnitBuilder.toHex(e.getASDU(), ""));
		}
		catch (final RuntimeException ex) {
			System.err.println(ex);
		}
	}
}

HTH, Boris

JourdanM, 2017-03-09:
Hello,

I tried it and it worked fine but is it possible to get the value instead of the ASDU ? How can I convert it ? 

Thanks a lot

calimero-project, 2017-03-09:
Use a DPT translator from package `tuwien.auto.calimero.dptxlator`, you probably need `DPTXlator2ByteFloat` with the datapoint type (DPT) 9.001.

JourdanM, 2017-03-10:
Okay I checked this class.
But my monitor class give me multiple different type of DPT, temperature, boolean etc. 
How can I manage it ?

JourdanM, 2017-03-10:
Do you have an example of translating from ASDU to DTP ?

calimero-project, 2017-03-11:
Calimero cannot know the exact DPTs you want in your setup. Either, you distinguish based on the datapoint group address, and create the desired translator. Or, you configure your datapoints up front to automate translation, like this:
``` 
DatapointModel<StateDP> datapoints = new DatapointMap<>();
datapoints.add(new StateDP(..., "my temp", 0, DPTXlator2ByteFloat.DPT_TEMPERATURE.getID()));
// add your other datapoints
...
// now, you can request a translator for a specific datapoint
TranslatorTypes.createTranslator(0, datapoints.get(...).getDPT());
```

JourdanM, 2017-03-15:
I tried something like this : 

`	private void print(final String svc, final ProcessEvent e) throws KNXFormatException
	{		
		DatapointModel<StateDP> datapoints = new DatapointMap<>();

		
		try {
			datapoints.add(new StateDP(new GroupAddress("1/1/3"), "my temp", 0, DPTXlator2ByteFloat.DPT_TEMPERATURE.getID()));
			System.out.println(LocalTime.now() + " " + e.getSourceAddr() + "->" + e.getDestination() + " " + svc + ": "
					+ e.getASDU());

			System.out.println(TranslatorTypes.createTranslator(0, datapoints.get(new GroupAddress("1/1/3")).getDPT()));
			
		}
		catch (final RuntimeException ex) {
			System.err.println(ex);
		} catch (KNXException e1) {
			// TODO Auto-generated catch block
			e1.printStackTrace();
		}
	}`

The result is this : 

09:39:51.733 1.1.9->1/1/3 write.ind: [B@2c8e27f7
DPT 9.001 [0.0 °C]

So I'm able to get the correct datapoint but the value is still 0.0. 
Am I doing something wrong to print my values ? 

Thanks a lot for the help

calimero-project, 2017-03-15:
You forgot to set the actual knx data you want to translate to a java type:

```
DPTXlator t = TranslatorTypes.createTranslator(...);
t.setData(e.getASDU());
System.out.println(t);
```

closed 2017-04-14 by calimero-project