calimero-project/calimero-core #46: Android connection error catch

ViorelOnica, 2017-08-11:
I'm trying to establish a KNX connection using calimero library.

While using a public static void main, it's working just fine (code example below), but I'm having troubles trying to connect in an Android application.

**Connection using public static void main (it works):**
`public class Main {
    private static final String remoteHost = "my_remote_ip_here";

      private static final String localHost = "my_local_ip_here";

      private static final String group = "0/0/1";
      private static final int knxServerPort = KNXnetIPConnection.DEFAULT_PORT;

    /**
     * @param args
     */
    public static void main(final String[] args) {
        KNXNetworkLink knxLink = null;
        ProcessCommunicator pc = null;
        System.out.println("This example shows how to establish a KNX connection "
            + "to a KNXnet/IP server.");

        try {
            final InetSocketAddress localEP = new InetSocketAddress(
                InetAddress.getByName(localHost), 0);
            final InetSocketAddress remoteEP = new InetSocketAddress(remoteHost, knxServerPort);

            System.out.println("Try connecting to " + remoteHost + " on port " + knxServerPort + "...");
            knxLink = new KNXNetworkLinkIP(KNXNetworkLinkIP.TUNNELING, localEP, remoteEP, true,
                TPSettings.TP1);

            System.out.println("Connection to " + remoteHost + " successfully established");

            knxLink.close();

            System.out.println("Connection got closed");
        }
        catch (final KNXException e) {

            System.out.println("Error connecting to " + remoteHost + ": " + e.getMessage());
        }

        catch (final InterruptedException e) {
            System.out.println("Connecting to " + remoteHost + " was interrupted, quit");
        }
        catch (final UnknownHostException e) {
            System.out.println("Host resolution for local endpoint " + localHost + " failed");
        }

    }
}`

**Connection in onCreate() (error catch):**

`public class MainActivity extends AppCompatActivity {
    private static final String remoteHost = "my_remote_ip_here";
    private static final String localHost = "my_local_ip_here";

    private static final String group = "0/0/1";
    private static final int knxServerPort = KNXnetIPConnection.DEFAULT_PORT;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);
        KNXNetworkLink knxLink = null;
        ProcessCommunicator pc = null;
        Log.d("asdf", "This example shows how to establish a KNX connection to a KNXnet/IP server.");

        try {
            final InetSocketAddress localEP = new InetSocketAddress(
                    InetAddress.getByName(localHost), 0);
            final InetSocketAddress remoteEP = new InetSocketAddress(remoteHost, knxServerPort);

            Log.d("asdf", "Try connecting to " + remoteHost + " on port " + knxServerPort + "...");
            knxLink = new KNXNetworkLinkIP(KNXNetworkLinkIP.TUNNELING, localEP, remoteEP, true, TPSettings.TP1);

            Log.d("asdf", "Connection to " + remoteHost + " successfully established");
            knxLink.close();

            Log.d("asdf", "Connection got closed");
        }
        catch (final KNXException e) {

            Log.d("asdf", "Error connecting to " + remoteHost + ": " + e.getMessage());
        }
        catch (final InterruptedException e) {
            Log.d("asdf", "Connecting to " + remoteHost + " was interrupted, quit");
        }
        catch (final UnknownHostException e) {
            Log.d("asdf", "Host resolution for local endpoint " + localHost + " failed");
        }

    }
}`

_knxLink = new KNXNetworkLinkIP(KNXNetworkLinkIP.TUNNELING, localEP, remoteEP, true,TPSettings.TP1);_
After executing this line, error is catched, by the following catch block:
`catch (final KNXException e) {
    Log.d("asdf","Error connecting to " + remoteHost + ": " + e.getMessage());
}`

I removed the real IPs in these examples, but everytime I'm trying to connect, I use legit IPs.
I'd appreciate some help. Thank you!

calimero-project, 2017-08-11:
Does your app has network permission in its manifest?

ViorelOnica, 2017-08-11:
thank you very much!

closed 2017-08-12 by ViorelOnica