calimero-project/calimero-core #52: ProcessCommunicator.readBool switches light

itineric, 2018-01-05:
Hello,

First of all, thanks for the lib, great work !
Im starting a projet and just testing stuff with the API and provided examples.
I have an issue I do not understand... I use processCommunicator.readBool(dpAddress) to get the current state of a switch group address. But when the read is performed, the light switches (off to on or the other way).
I cannot understand why a read changes some value ?!
How can I get the state of GA to initialize the states (get if the light is on or off) ?

Regards,

Eric

bmalinowsky, 2018-01-05:
readXXX is the right command for read-only stuff. The read command does not contain any state of that group address, so I think the described behavior comes from somewhere else.

Do you have a chance to use a monitor while reading? For example, the ETS monitor, or the calimero ProcComm tool or GUI. (If you have a second KNX interface you can start a _network/bus_ monitor on it, on the same interface you can only use the _group_ monitor in parallel.)

It could be that some knx device issues a group write as response to the read? Or a read response is interpreted as write? This does not explain the toggle effect, though...

itineric, 2018-01-06:
Thanks, I effectively have the same issue when reading the value of that particular GA using the group monitor.
Still havent found why but looking into it but not a Calimero issue. Sorry.

closed 2018-01-29 by bmalinowsky