calimero-project/calimero-core #56: Better using 2.3 Release or current 2.4 master for new Projects

lewie, 2018-01-20:
@bmalinowsky, is it foreseeable when version 2.4 reached the beta level? I think about what would be better to use currently,  2.3 Release or current 2.4 master for a new Project.

bmalinowsky, 2018-01-29:
It will be in the short-term future, there aren't big changes I can think of (probably removal of deprecated code, moving the KNX PB-mode link procedure, etc.).
I will certainly release it before committing the module-infos and other j9 stuff. OTOH v2.3 will stay _as-is_, and also not receive any updates (unless _really_ serious).

lewie, 2018-01-31:
I see that correct [this fix](https://github.com/calimero-project/calimero-core/issues/57#issue-291929017) will only infuse in the 2.4 branch.

Is so, we should use 2.4 master.

closed 2018-01-31 by lewie

bmalinowsky, 2018-01-31:
Well, the fix is in master now. 

For 2.3, I don't know how many users are actually affected by that. 
Obviously, the increased resource allocation over time is not nice.
For example: even assuming 2 link creations initially fail per hour (ctor throws, for whatever reason), one can probably run it for a month (48 * 30) even on small platforms before running into memory problems?

muenning, 2018-01-31:
Well, in my case the memory was not the problem but the 8192 thread limit. With OpenHAB 2.x this limit was reached within 12 hours. As this fix seems simple, IMHO backporting it to v2.3 and using that modified (and otherwise static) unofficial version in OH should not be a big deal and generally an one-time effort.

bmalinowsky, 2018-01-31:
@muenning yes, it's an easy fix. If the openhab devs think it's useful, i can update v2.3, too.

Your numbers indicate that you basically run a busy wait for server connection (?)

muenning, 2018-01-31:
@bmalinowsky these numbers come from a busy OpenHAB system where the problem starts with dropping and reestablishing connections to a MDT KNX router. The calimero bug may be triggered when the connection slots are exhausted but I think this may be a second bug triggered by another bug. It seems to me that there is a bug in the re-connection handling in OH which could be the trigger. For details you may look at openhab/openhab1-addons#5440.

sjka, 2018-02-01:
Just to notice: It's great that the fix for #57 will make it into 2.4, but from my point of view there is no need to create a 2.3 patch nor is this a reason to switch to 2.4-SNAPSHOT immediately - there is an easy workaround by using the other constructor (see e.g.  [openhab/openhab2-addons#2323](https://github.com/openhab/openhab2-addons/pull/2323/files#diff-9d9efb354f43eb37549a28181d1fd01eR87))