calimero-project/calimero-core #58: Can write data, but can't read data

victor-iancu, 2018-02-05:
I've tried out today this example from calimero-introduction-repo, version 2.4:

```java
import java.net.InetAddress;
import java.net.InetSocketAddress;
import java.net.UnknownHostException;

import tuwien.auto.calimero.GroupAddress;
import tuwien.auto.calimero.KNXException;
import tuwien.auto.calimero.link.KNXNetworkLink;
import tuwien.auto.calimero.link.KNXNetworkLinkIP;
import tuwien.auto.calimero.link.medium.TPSettings;
import tuwien.auto.calimero.process.ProcessCommunicator;
import tuwien.auto.calimero.process.ProcessCommunicatorImpl;

/**
 * Example of Calimero process communication, we read (and write) a boolean datapoint in the KNX network. By default,
 * this example will not change any datapoint value in the network.
 */
public class ProcessCommunication {
    // Address of your KNXnet/IP server. Replace the IP host or address as necessary.
    private static final String remoteHost = "192.168.1.84";

    // We will read a boolean from the KNX datapoint with this group address, replace the address as necessary.
    // Make sure this datapoint exists, otherwise you will get a read timeout!
    private static final String group = "0/5/6";

    public static void main(final String[] args) {
        final InetSocketAddress remote = new InetSocketAddress(remoteHost, 3671);
        // Create our network link, and pass it to a process communicator
        try (KNXNetworkLink knxLink = KNXNetworkLinkIP.newTunnelingLink(null, remote, false, TPSettings.TP1);
             ProcessCommunicator pc = new ProcessCommunicatorImpl(knxLink)) {

            System.out.println("read boolean value from datapoint " + group);
            final boolean value = pc.readBool(new GroupAddress(group));
            System.out.println("datapoint " + group + " value = " + value);

//            final boolean value = true;
            // Uncomment the next line, if you want to write back the same value to the KNX network
//            pc.write(new GroupAddress(group), value);
        } catch (KNXException | InterruptedException e) {
            System.out.println("Error accessing KNX datapoint: " + e.getMessage());
        }
    }
}
```

This code works fine when I'm only writing to the KNX system: `pc.write(new GroupAddress(group), value)`, but when I try to read a value: `pc.readBool(new GroupAddress(group))` from the same group address, I get a read timeout exception.

I've also searched through other closed issues (#41, #21), but nothing from there seems to work.
I'm building an running my code from Mac OS 10.13.3 through IntelIJ IDEA.

My KNX system is available through an IP-router on a local address.
The datapoint I've tested on is my room's light system.
The discover KNX/IP server example prints this result:

```
en0 /192.168.1.163 <= 192.168.1.84:3671 (IPv4 UDP) "KNX / IP-Router REG-K"
	KNX address 1.1.0
	KNX medium TP1
	installation 0 - project 0 (ID 0)
	routing multicast address 224.0.23.12
	MAC address myMacAddress
	S/N mySN
	Core (v1)
	Device Management (v1)
	Tunneling (v1)
	Routing (v1)
```

Any suggestions on what I should do? 
Thanks!

Please note: I'm new to the whole KNX environment, just started learning about it 2 days ago :D

bmalinowsky, 2018-02-06:
I would first make sure the group address you try reading from actually responds to read requests. This is not always the case (it's configured). 
You can also run the group monitor example, and see what messages come in (i.e., what addresses) when you switch the lights etc.

victor-iancu, 2018-02-06:
I've contacted the KNX system's admin and it seems I've had the wrong group address to read from :(
Now it's all working fine.
Thanks for help!

closed 2018-02-06 by victor-iancu