calimero-project/calimero-core #59: library import

ViorelOnica, 2018-03-10:
Hi there, 

I used this library 7 months ago, and I resumed the project (android), but classes are not recognized anymore.. 
I placed " implementation 'com.github.calimero-project:calimero-core:v2.3' " in my gradle, "maven { url 'https://jitpack.io' } " aswell. 

If I'll get through this problem, I'd like to know if there's any way to get notified whenever the knxconnection goes down without running a second thread with a infinite loop in it that checks the connection.. 

I'd appreciate your help.. :)

calimero-project, 2018-03-11:
The gradle version part does not start with a 'v', remove that.

Every connection implementation has a method to add a `tuwien.auto.calimero.KNXListener`, and every link implementation has a method to add a `tuwien.auto.calimero.link.LinkListener`. Both will provide you with a notification if the connection/link got closed.

ViorelOnica, 2018-03-13:
Hello, thank you for your answer !

It's working, but I tested it deactivating mobile data and the linkClosed event is triggered after 30-45 seconds after turning mobile data off.. ain't 30 seconds too much? 

DataHolder.knxLink.addLinkListener(new NetworkLinkListener() {
                @Override
                public void confirmation(FrameEvent e) {
                }

                @Override
                public void indication(FrameEvent e) {
                }

                @Override
                public void linkClosed(CloseEvent e) {
                         //what to do on losing connection
                }
            });

calimero-project, 2018-03-14:
The notification will be sent when the link actually got closed; if the link is idle and does not do any communication (and the socket is not closed by the OS), then there can be a certain time delay. 

On Android, if you want an immediate notification about changes to your connection, you probably should use `android.net.ConnectivityManager. CONNECTIVITY_ACTION` or something like that.

closed 2018-03-22 by calimero-project