calimero-project/calimero-core #64: calimero dimming read and write

ViorelOnica, 2018-05-06:
Hello,
I have 2 problems:
1) Reading switch (1bit boolean) and dimming (1byte unsigned int) datapoints. 
In order to do that, I used these 2 functions below, but "catch" occurs.

[image]

Below you can see the objects I created and the logs read attempts.
[image]

Is it about bad objects configuration or am I using wrong the library methods ?

2) I have a method that gets triggered whenever knx connection gets lost:
@Override
public void linkClosed(CloseEvent e) { //code to execute }

But I tried the following scenario: turned off mobile data and this function gets triggered after 30-45 seconds after losing the internet connection. 
You told me in another issue post that what I'm seeking for is an idle state trigger over the knxlink.. I failed finding the proper function that takes care of that, can you give me a concrete code example, please ?

I'd really appreciate your help !

calimero-project, 2018-05-08:
The way you use the methods is fine.

What exactly is the error (output of `e.printStackTrace()`)?

Re link: if your goal is to automatically reconnect and/or configure that in more detail, you can use Connector (`tuwien.auto.calimero.link.Connector`). Even if the network connection was down (and/or the link got closed), it will re-establish a link.

ViorelOnica, 2018-05-08:
[image]

This exception is threw after calling any of the 2 functions (readBoolean, readDimming) ..;\

Concerning the knxlink: 
Yes, I want to automatically reconnect, but there's a green circle in the application indicating that the connection's up and I want to make it red whenever it goes idle ( I'm using the linkClosed method and it's getting red only after 30-45 seconds after losing the mobile data )

calimero-project, 2018-05-08:
This exception means that the remote endpoint (the one that should answer for that group address) is not answering, therefore, a timeout exception is thrown after the specified process communication timeout.

ViorelOnica, 2018-05-08:
hmm, okay, thanks.. I'll do some research

ViorelOnica, 2018-05-10:
well, I'm not getting timeout exception anymore and I'm getting a response aswell

[image]

But this is the new exception I get

[image]