calimero-project/calimero-core #66: 1 byte datapoint read

ViorelOnica, 2018-05-31:
Hello,

I asked some questions concerning datapoints reading in some previous issues, but I still encounter some problems.. ;\

The exception I get when I call readControl is "APDU response length 3 bytes, expected 2 to 2".
And it's weird.. because in the knx control panel (the one where I can create and access objects), my object appears to be "1 byte unsigned integer".

bmalinowsky, 2018-05-31:
A 1 byte unsigned integer is not a control DPT, but an 8-bit (i.e. 1 byte) unsigned DPT. You read it with `readUnsigned` and probably `SCALING` to get values between 0 - 100.

ViorelOnica, 2018-06-01:
succeded, thank you very much !

ViorelOnica, 2018-06-01:
"This format scales the 8 Bit unsigned value range from 0 to 100. "
My object range is 0-255.. Is there any way I can read the actual value (0-255) or I have to use interpolation ?

For example, the current value was 54, the method returns 21 ( because (54*100)/255 = 21 )

bmalinowsky, 2018-06-01:
use `UNSCALED`

closed 2018-06-06 by bmalinowsky