calimero-project/calimero-core #67: Java 9

steveohara, 2018-06-06:
Can I ask why the library requires Java 9?
I presume that a lot of people will want to incorporate the library into their own applications and creating a dependency on Java 9 will make that much more difficult. (I can't tell you how many complaints I got from people when I dared to try and uplift [j2mod ](https://github.com/steveohara/j2mod)to Java 7!!)
Java 8 would seem reasonable but 9 is very new for most industrial vendors.

steveohara, 2018-06-06:
The issues we have come across are that some users cannot update the runtime JVM - it is pre-installed or simply not something they can influence.
Java 8 is fine, just a bit miss-leading on your read me that says it requires Java SE 9

closed 2018-06-06 by steveohara

bmalinowsky, 2018-06-06:
Ah ok. The readme got updated with moving to 2.5 snapshots couple weeks ago. If you look at the [2.4 branch](https://github.com/calimero-project/calimero-core/tree/release/2.4) the requirements are different.

petero-dk, 2019-10-09:
Sorry to add a comment for a closed issue. But I just found this during a search, and I was in the same situation. I needed an updated Calimero but for a Java 8 project, which is why I managed to make a (transpiler, dunno what is is called) that converts a Java 9 / Java 10 or Java 11 jar to a Java 8. Using Google R8 and Dex2Jar.

https://github.com/petero-dk/javacrosscompiler

I am working on getting a direct download link, but for now Snapshot builds are instantly available by clicking the Azure pipelines status button and selecting "Artifacts"