calimero-project/calimero-core #75: New Translators for extended color DPTs

lewie, 2018-11-08:
Hello @bmalinowsky,

```
/*
 * 7.600  DPT_Absolute_Colour_Temperature [K] (TW) ------------>>> ETS knows, calimero doesn't
 * 232.600 DPT_Colour_RGB, values: 0 0 0...255 255 255, r g b->>> ETS knows, calimero too! 
 * 242.600 DPT_Colour_xyY
 * 243.600 DPT_Colour_Transition_xyY
 * 249.600 DPT_Brightness_Colour_Temperature_Transition ------>>> ETS knows, calimero doesn't
 * 250.600 DPT_Brightness_Colour_Temperature_Control --------->>> ETS knows, calimero doesn't
 * 251.600 DPT_Colour_RGBW ----------------------------------->>> ETS knows, calimero doesn't
 * 252.600 DPT_Relative_Control_RGBW
 * 253.600 DPT_Relative_Control_xyY
 * 254.600 DPT_Relative_Control_RGB
 */
```

There is several Hardware that uses these 2-byte, 3-byte and 6-byte color specific DTPs.
Jung DALI Gateway TW
Gira KNX DALI-Gateway
MDT: LED Controller AKD, DaliControl IP Gateway DALI64, DaliControl Gateway DALI16/32

RGBW and other DTP information/definitions: https://www.mdt.de/download/MDT_THB_DaliControl_Gateway_03.pdf

What do you think, isn't it time to integrate the new color DTPs? Eventually for calimero 2.4 too?

Thank You!

bmalinowsky, 2018-11-09:
That makes sense. 

v2.4 is currently rc2, so additions are by default in master. But since new translators won't affect existing behavior, I can backport them into 2.4 if the translators have been tested before sufficiently. I can't test myself because I don't have HW with those DPTs.

I created a feature branch based on master [here](https://github.com/calimero-project/calimero-core/tree/feat/color-xlators) to track progress.
- [x] 7.600 DPT_Absolute_Colour_Temperature, **implemented** in dc7de7c18fc6983aa1e670cd26cc81a508049e08.
- [x] 232.600 DPT_Colour_RGB, **existed already**
- [ ] 242.600 DPT_Colour_xyY
- [ ] 243.600 DPT_Colour_Transition_xyY
- [x] 249.600 DPT_Brightness_Colour_Temperature_Transition, **implemented in feature branch**
- [ ] 250.600 DPT_Brightness_Colour_Temperature_Control
- [x] 251.600 DPT_Colour_RGBW, **implemented in feature branch**
- [ ] 252.600 DPT_Relative_Control_RGBW
- [ ] 253.600 DPT_Relative_Control_xyY
- [ ] 254.600 DPT_Relative_Control_RGB

calimero-project, 2018-11-17:
@lewie do you by any chance have access to HW which uses those dpts, and try them out?

lewie, 2018-11-25:
First of all, best thanks for this patch!!
It takes me some weeks, then I have suitable hardware and will test!

Have tested it so far with ETS, simulating telegrams sending to a prepared openHAB and calimero worked/identified as as expected.

calimero-project, 2020-08-23:
All the color dpts are now implemented (some fine-tuning might be required).

closed 2020-08-23 by calimero-project