calimero-project/calimero-core #77: T_Disconnect telegrams triples bus activity while programming

lewie, 2018-11-18:
Hello @bmalinowsky,
we have a bad effect while programming devices with the ETS. 
Please, can you put us on the right track al little? ;-)
T_Disconnect telegrams triples bus activity while programming.

Discussion: https://github.com/openhab/openhab2-addons/issues/4238

Thank You!

bmalinowsky, 2018-11-20:
The disconnects in itself are harmless, but IIRC were meant for a more esoteric condition where client-initiated connections got already closed or addressed by the wrong sender. They become a nuisance in a situation with >1 mgmt clients (of which ETS is one) together in the same mcast group.

I leave this issue open to address it with a less traffic-heavy/polluting send approach.

JoveToo, 2018-11-20:
@bmalinowsky The result of the extra traffic on the bus was a consistent firmware download failure, not quite harmless.

lewie, 2018-11-21:
I can really confirm the observation. Depending on the hardware device, the T_Disconnect telegrams bother so much that in many cases, the firmware download fails.

bmalinowsky, 2018-11-22:
@lewie what happens if you check the compatibility checkbox for slower bus communication in ets? Same result?

commit 65856723 (2018-11-22): Keep layer 4 disconnects to a minimum (fixes #77)

If a link got a device address assigned, the transport layer will use
that address to decide whether to send a disconnect.
commit 2942c927 (2019-01-06): Keep layer 4 disconnects to a minimum (fixes #77)

If a link got a device address assigned, the transport layer will use
that address to decide whether to send a disconnect.
closed 2018-11-22 by bmalinowsky