calimero-project/calimero-core #81: KNX Data Secure support

nanosonde, 2019-01-08:
Do you also plan to support for KNX Data Secure?

I also wonder if you are going to implement the feature "security proxy" for the calimero server?
This would enable the use of KNX RF Data Secure components with a normal KNX TP installation without KNX TP Data Secure components.

bmalinowsky, 2019-01-10:
Eventually, but there is no specific time plan for now. The first thing would certainly be to support configuration etc. I did not look into any HW for (raw) I/O, but as you say, RF is a prime candidate where people want to use it first.

The secure proxy (from a server perspective) is supported, but is client <--> server. 
So I think you refer to message exchange between knx subnets?

* If you use an RF data secure stick and an TP1 stick, you should get what you want, as one can do now already. Maybe additional settings are required for RF secure in the server config (I don't know, never tried).
* If you want raw I/O over RF w/ data secure processing in calimero, then no (as this requires the data secure stack implementation).

bmalinowsky, 2019-01-12:
Regarding data secure stick, that was maybe a bit unclear: I was referring to the general setup which already works (e.g., connecting an RF and TP1 subnet); when a suitable data secure usb stick will be commercially available I don't know. The current product landscape is quite sad. 

But if such an adapter is configurable (links/keys etc.) by ETS using a secure installation, I strongly assume it provides decrypted frames. Otherwise, it would be somewhat useless IMO.
(This is in contrast to what I was referring to as 'raw' I/O for which a data secure _software_ stack is actually interesting: an RF adapter which happily sends out an already prepared secure frame without caring about keys, encryption, mac etc.)

nanosonde, 2020-05-06:
Hi!
There is a recent post saying that KNX AN192
has been publish to the manufacturers:
https://knx-user-forum.de/forum/öffentlicher-bereich/knx-eib-forum/1440881-knx-rf-secure?p=1441214#post1441214

If I understand it correctly then it is supposed to contain the security proxy spec.
Do you habe access to the latest KNX specs?

Is there some docu available on how to use your latest data secure addition (e.g. KNX Server)?

nanosonde, 2020-05-12:
According to [this](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=3&cad=rja&uact=8&ved=2ahUKEwiznbj00q3pAhXOyKQKHVPiAe0QFjACegQIAhAB&url=https%3A%2F%2Fmedia.knx.org%2Ffeed%2Ffile%2F1032&usg=AOvVaw0sC6px_Vzo6K1lifNJztC9) PDF. AN192 is called "Coupler Security Extensions".

bmalinowsky, 2021-05-13:
Closing since Data Secure got integrated.

closed 2021-05-13 by bmalinowsky