calimero-project/calimero-core #84: Discoverer

ViorelOnica, 2019-03-23:
Hello !

I used the code below to discover my local knx devices (when I'm connected to wi-fi). I tested a lot of devices and it worked, all were listed, excepting one knx device (Gewiss GWA9705) that wasn't listed, altought it was configured properly and it was discovered somehow by ETS Software. 
My best guess was that the discoverer doesn't work on IP interface devices, just on IP router devices. 
Is there any way I can discover ALL my knx devices, regardless its type ?

final Discoverer discoverer = new Discoverer(null, 0, false, true);
discoverer.startSearch(3, true);
for(int i=0;i<discoverer.getSearchResponses().length;i++)
	Log.d("asdf","discovery: "+discoverer.getSearchResponses()[i].getControlEndpoint().getAddress());

bmalinowsky, 2019-03-25:
Interfaces and routers are detected and answer in the same way. Maybe check if you see any stack trace or error in the log for that particular case?

Can you otherwise communicate fine with the Gewiss device (like groupmon etc.)?

ViorelOnica, 2019-03-28:
No logs.
Yes, I can establish connection with that device, groupmonitor works, the only problem is it's not discovered. :\

calimero-project, 2020-06-16:
Please reopen if there is new information that helps.

closed 2020-06-16 by calimero-project