calimero-project/calimero-core #88: maximum send attempts, no service acknowledgment received

lewie, 2019-08-04:
@bmalinowsky,
do you have any idea about that:
https://github.com/openhab/openhab2-addons/issues/5861#issue-472566348

Thank you for your help in advance...

PBrou, 2019-08-27:
Suffering from this issue on raspberry Pi3+, if you need some testing, please let me know.
Kr
PBrou

bmalinowsky, 2019-09-03:
@PBrou  does a confirmation even show on the network, i.e., is it sent by your router? If not, there is nothing much to do other than to close the connection. If a confirmation was sent, then it could indicate a protocol problem. (If your router supports telnet, you could check if there is further information.)

PBrou, 2019-09-06:
@bmalinowsky ,
It is not clear for me if there is a correlation between the large set of "no confirmation reply received" KNX telegrams piling up and the eventually crashing/stopping/disconnect with the KNX bus.
I checked my KNX bus and I did find out that for an alarm event on a movement sensor with 1-bit object and also a new CO2/Humidity/temp sensor with 2-byte value objects there is/was in fact no receive confirmation as there is only this single object in the KNX group. At some moment I saw also a lot of LL-BUSY on my telegrams.
Due to this (only one item in KNX group) these KNX sensors are/were constantly resending (2 extra times) which as such is normal KNX behavior but slightly overloading needlessly my KNX bus.
For the movement sensor event I fixed it by just connecting it into a non-used input of an actor, but for the 2-byte value I don't know yet how to do this.
@lewie ,
I have here a question for openhab2.x - sorry for hijacking this thread for this - is there a way that openhab can act as consumer (like an actor KNX device) of such value broadcast group telegram and provide the receive confirmation?

I hope this make some sense.. I am using the - inferior (but cheaper) GIRA IP Interface.. (not the router) and thus using tunnel mode. With OH2.4 (and before the new CO2/Hum/Temp sensor) previously on Mac mini (before migration to Pi3B+ my KNX connection was super stable.
I'll continue to do some tests, removing some elements to narrow down the problem.

bmalinowsky, 2019-09-10:
(`15.15.255` is usually not used as a default, whether in calimero or knx.)

There is a correlation between .con and closing the connection wrt that a .con is required by the protocol (so the server has to send it). Calimero won't close the connection based on that, but it's an indicator that something ran afoul.

The missing ack based on which calimero closes the connection, is the acknowledge of a tunneled frame by the server, and solely related to the tunneling protocol (not to any ack on the knx bus).

From what you describe I currently would suspect a timing issue, which also became observable over the tunneling connection. 

If you can deterministically reproduce the behavior, you can check for the (missing) confirmation over IP using wireshark (type "knx" in the display filter). This would clarify whether the confirmation is lost due to timing (too late), or not sent at all.

PBrou, 2019-10-06:
I finally have my knx bus traffic logging + wireshark + port mirroring on my switch.
In this very first analysis I don't think there are real issues on the KNX IP traffic between my Pi3B+ and the IP Gateway itself.
However I do see some (strange) double requests sometimes. Openhab (2.5.0 Build 1711) seems to trigger much too soon a second try on some KNX group read!?
[image]
It also looks like there is a tunnel check connection done every 1s - probably cannot harm?
[image]
I will do some more testing with KNX Bus logging + Calimero debug level in openhab + wireshark.

bmalinowsky, 2019-10-07:
The repetition after 1 second is required by the protocol (and 1 second in a LAN is a long time to wait for an ACK). You can see the "normal" timing if your KNX server answers it does it quite quickly, there is a factor 10^3 difference. 
Another strange thing is that the server ignores a request, but then sends a .con twice, even though the first gets immediately ACKed by the client. Not normal behavior.

Regarding tunnel check connection: you mean the connection-state request? That's done every minute, not second.

PBrou, 2019-10-27:
Please find some more test results from a few weeks ago.
In my case it looks like sometimes there are one or two KNX request without an ACK on the bus.
I have the impression that the software kind of looses control/track of the situation after such situation ( some error stack not cleaned, some error situation that stays pending somewhere ).
The two requests without ACK lead to a situation with mixed, interweaving of two parallel retries..

### View from the KNX BUS
**All normal first..**
21:52:47.084	S=0	L	1.1.19	Keuken -	2/1/0	Keukeneiland	6	Write	1 bit	$00	LL-ACK
21:52:47.194	S=2	A	1.1.3	Lichten.	2/3/0	Status Keuken	6	Write	1 bit	$00	LL-ACK21:53:18.970	S=4	L	15.15.255	Openhab Server	10/1/0	Vochtigheid Badkamer	6	Read	-	LL-ACK
21:53:37.317	S=6	L	1.1.39	Sensor Vochtigheid, Temp, CO2	10/1/0	Vochtigheid Badkamer	6	Response	2 byte	15 36 | 53,36	LL-ACK

**The following read does not show an ACK on the bus (?)**
21:54:17.314	S=0	L	15.15.255	Openhab Server	10/0/0	Temperatuur Badkamer	6	Read	-
**We'll see retries for this in the openhab log file.
Meanwhile a 2nd read is launch (while the other is still retrying)
This one also on the bus without ACK**
21:54:57.312	S=4	L	15.15.255	Openhab Server	10/2/0	CO2 Badkamer	6	Read	-
**Then back to normal (just status messages)**
21:55:26.574	S=0	L	1.1.2	Rolluiken Garage; Sturing Verwarming; Lichten Toilet+Verluchting+Lichten buiten.	2/3/9	Status Licht Toilet Boven	6	Write	1 bit	$00
21:55:26.584	S=2	L	1.1.3	Lichten.	2/3/2	Status Wasplaats Licht 1	6	Write	1 bit	$00

**And then back too with a normal read situation with ACK on the bus**
21:58:19.072	S=4	L	15.15.255	Openhab Server	10/1/0	Vochtigheid Badkamer	6	Read	-	LL-ACK
21:58:19.102	S=6	L	1.1.39	Sensor Vochtigheid, Temp, CO2	10/1/0	Vochtigheid Badkamer	6	Response	2 byte	15 34 | 53,28	LL-ACK

**And now a 10/0/0 back normal with an ACK**
21:58:22.346	S=0	L	15.15.255	Openhab Server	10/0/0	Temperatuur Badkamer	6	Read	-	LL-ACK
21:58:22.386	S=2	L	1.1.39	Sensor Vochtigheid, Temp, CO2	10/0/0	Temperatuur Badkamer	6	Response	2 byte	0C 88 | 23,2	LL-ACK
21:58:25.641	S=4	L	15.15.255	Openhab Server	10/2/0	CO2 Badkamer	6	Read	-	LL-ACK
21:58:25.671	S=6	L	1.1.39	Sensor Vochtigheid, Temp, CO2	10/2/0	CO2 Badkamer	6	Response	2 byte	2E D6 | 560	LL-ACK

### View on the KNX IP traffic (PI <-> IP Interface tunnel mode)

Line 4420+4430: there seems to be a double response value on the IP KNX level - normal?
Line 4423, 4432,4539,.: the multiple retries on the 10/0/0 groupRead request

[image]

### View in openhab log file with calimero tracing on

2019-10-08 21:52:47.163 [TRACE] [.internal.handler.DeviceThingHandler] - onGroupWrite Thing 'knx:device:b063da47' processes a GroupValueWrite telegram for destination '2/1/0' for channel 'knx:device:b063da47:Light_GF_Kitchen_TL'
2019-10-08 21:52:47.169 [TRACE] [.internal.handler.DeviceThingHandler] - onGroupWrite isControl
2019-10-08 21:52:47.177 [TRACE] [g.knx.internal.dpt.KNXCoreTypeMapper] - toType datapoint DPT = 1.001
2019-10-08 21:52:47.183 [TRACE] [.knx.internal.channel.KNXChannelType] - getCommandSpec testing Keys '[ga]' for command 'OFF'
2019-10-08 21:52:47.192 [TRACE] [.knx.internal.channel.KNXChannelType] - getCommandSpec key 'ga' uses expectedTypeClass 'class org.eclipse.smarthome.core.library.types.OnOffType' witch isInstance for command 'OFF' and dpt '1.001'
2019-10-08 21:52:47.201 [TRACE] [.internal.handler.DeviceThingHandler] - rememberRespondingSpec handled commandSpec for '2/1/0' size '1' added 'true'
2019-10-08 21:52:47.208 [TRACE] [g.knx.internal.dpt.KNXCoreTypeMapper] - toType datapoint DPT = 1.001
2019-10-08 21:52:47.215 [TRACE] [.internal.handler.DeviceThingHandler] - processDataReceived postCommand new value '[0]' for GA 'null'
2019-10-08 21:52:47.233 [TRACE] [Xnet/IP Tunneling 192.168.1.249:3671] - received request seq 35 (channel 67) cEMI 29 00 b8 e0 11 03 13 00 01 00 80
2019-10-08 21:52:47.241 [TRACE] [nx.internal.client.AbstractKNXClient] - Received a Group Write telegram from '1.1.3' to '2/3/0' with value '[0]'
2019-10-08 21:52:47.241 [DEBUG] [calimero.link.192.168.1.249:3671    ] - indication 1.1.3->2/3/0 L_Data.ind, urgent priority hop count 6, tpdu 00 80
2019-10-08 21:52:48.943 [TRACE] [Xnet/IP Tunneling 192.168.1.249:3671] - sending connection state request, attempt 1
2019-10-08 21:53:18.962 [TRACE] [nx.internal.client.AbstractKNXClient] - Sending a Group Read Request telegram for 10/1/0
2019-10-08 21:53:18.972 [DEBUG] [calimero.link.192.168.1.249:3671    ] - send (wait for confirmation) 15.15.255->10/1/0 L_Data.req, low priority hop count 6 repeat, tpdu 00 00
2019-10-08 21:53:18.981 [TRACE] [Xnet/IP Tunneling 192.168.1.249:3671] - sending cEMI frame seq 9, WaitForCon, attempt 1 (channel 67) 06 10 04 20 00 15 04 43 09 00 11 00 bc e0 ff ff 51 00 01 00 00
2019-10-08 21:53:18.988 [TRACE] [Xnet/IP Tunneling 192.168.1.249:3671] - received service ack 9 from /192.168.1.249:3671 (channel 67)
2019-10-08 21:53:19.032 [DEBUG] [Xnet/IP Tunneling 192.168.1.249:3671] - received request seq 36 (channel 67) cEMI L-Data.con 15.15.255->10/1/0
2019-10-08 21:53:19.041 [DEBUG] [calimero.link.192.168.1.249:3671    ] - confirmation of 10/1/0
2019-10-08 21:53:19.046 [TRACE] [calimero.link.192.168.1.249:3671    ] - send 15.15.255->10/1/0 succeeded
2019-10-08 21:53:19.052 [TRACE] [ess.communication 192.168.1.249:3671] - sent group read request to 10/1/0
[... cut ...]
**2019-10-08 21:58:22.103 [WARN ] [Xnet/IP Tunneling 192.168.1.249:3671] - response timeout waiting for confirmation**
tuwien.auto.calimero.KNXTimeoutException: no confirmation reply received for 15.15.255->10/1/0 L_Data.req, low priority hop count 6 repeat, tpdu 00 00

**2019-10-08 21:58:22.161 [DEBUG] [nx.internal.client.AbstractKNXClient] - Could not read value for datapoint 10/1/0: no confirmation reply received for 15.15.255->10/1/0 L_Data.req, low priority hop count 6 repeat, tpdu 00 00. Going to retry.**

petero-dk, 2020-01-03:
@PBrou I am also a KNX in openHAB user (although I am going the serial device route and not tcp/ip) I thought I would just let you know about two things:
1. When you paste log files in github issues they are much easier to read if you enclose them in code tags ``` (blank line, code tags, log content, code tags, blank line) then it will look like this

```
this can be a really long log line, but it will not wrap, never, not ever, never ever ever ever ever ever ever ever ever ever ever wrap. No never, not ever. Still wont.
```

2. The calimero version used in the knx binding is actually quite old (comparativly speaking, in calendar time it is not, but @bmalinowsky is doing a lot of work :-)). It is fixed at version calimero 2.4 because that was the last version to support Java 8. So none of the latest developments in calimero is reaching openHAB until (probably) openHAB 3

petero-dk, 2020-01-03:
This is the very latest knx binding for openHAB with the very latest calimaro source (back compiled to java8). I do not know if it will make a difference for you.

[org.openhab.binding.knx-2.5.0-SNAPSHOT.jar.zip](https://github.com/calimero-project/calimero-core/files/4020904/org.openhab.binding.knx-2.5.0-SNAPSHOT.jar.zip)

villaRob, 2021-01-08:
Hi all,
any news on this ? I'm facing the same problem also with the stable openHAB 3 and latest KNX binding.

I'm available for testing or if further information is needed.

Basically, the issue is the same I had with OH2 (reason that block me to leave openhab 1.8.3 as my working enviroment)

This is an extract of openhab log:

```
2021-01-07 18:33:14.580 [DEBUG] [calimero.link.192.168.7.107:3671    ] - indication 0.1.25->0/2/8 L_Data.ind, low priority hop count 6, tpdu 00 80 00 2e
2021-01-07 18:33:14.610 [TRACE] [Xnet/IP Tunneling 192.168.7.107:3671] - received request seq 115 (channel 5) cEMI 29 00 bc e0 01 19 02 09 03 00 80 00 12
2021-01-07 18:33:14.611 [DEBUG] [calimero.link.192.168.7.107:3671    ] - indication 0.1.25->0/2/9 L_Data.ind, low priority hop count 6, tpdu 00 80 00 12
2021-01-07 18:33:14.950 [DEBUG] [calimero.link.192.168.7.107:3671    ] - send (wait for confirmation) 0.1.242->6/1/1 L_Data.req, low priority hop count 6 repeat, tpdu 00 80
2021-01-07 18:33:14.951 [TRACE] [Xnet/IP Tunneling 192.168.7.107:3671] - sending cEMI frame seq 3, WaitForCon, attempt 1 (channel 5) 06 10 04 20 00 15 04 05 03 00 11 00 bc e0 01 f2 31 01 01 00 80
2021-01-07 18:33:14.955 [TRACE] [Xnet/IP Tunneling 192.168.7.107:3671] - received service ack 3 from /192.168.7.107:3671 (channel 5)
2021-01-07 18:33:14.959 [TRACE] [Xnet/IP Tunneling 192.168.7.107:3671] - received request seq 116 (channel 5) cEMI 29 00 bc e0 01 f2 31 01 01 00 80
2021-01-07 18:33:14.959 [DEBUG] [calimero.link.192.168.7.107:3671    ] - indication 0.1.242->6/1/1 L_Data.ind, low priority hop count 6, tpdu 00 80
2021-01-07 18:33:15.114 [TRACE] [Xnet/IP Tunneling 192.168.7.107:3671] - received request seq 117 (channel 5) cEMI 29 00 bc e0 01 3b 31 02 01 00 80
2021-01-07 18:33:15.115 [DEBUG] [calimero.link.192.168.7.107:3671    ] - indication 0.1.59->6/1/2 L_Data.ind, low priority hop count 6, tpdu 00 80
2021-01-07 18:33:17.103 [TRACE] [Xnet/IP Tunneling 192.168.7.107:3671] - received request seq 118 (channel 5) cEMI 29 00 bc e0 01 42 32 0e 05 00 80 3f 8e 14 7b
2021-01-07 18:33:17.104 [DEBUG] [calimero.link.192.168.7.107:3671    ] - indication 0.1.66->6/2/14 L_Data.ind, low priority hop count 6, tpdu 00 80 3f 8e 14 7b
2021-01-07 18:33:17.958 [WARN ] [Xnet/IP Tunneling 192.168.7.107:3671] - response timeout waiting for confirmation
tuwien.auto.calimero.KNXTimeoutException: no confirmation reply received for 0.1.242->6/1/1 L_Data.req, low priority hop count 6 repeat, tpdu 00 80
	at tuwien.auto.calimero.knxnetip.ClientConnection.doExtraBlockingModes(ClientConnection.java:244) [bundleFile:?]
	at tuwien.auto.calimero.knxnetip.ConnectionBase.send(ConnectionBase.java:258) [bundleFile:?]
	at tuwien.auto.calimero.knxnetip.KNXnetIPTunnel.send(KNXnetIPTunnel.java:177) [bundleFile:?]
	at tuwien.auto.calimero.link.KNXNetworkLinkIP.onSend(KNXNetworkLinkIP.java:264) [bundleFile:?]
	at tuwien.auto.calimero.link.AbstractLink.send(AbstractLink.java:332) [bundleFile:?]
	at tuwien.auto.calimero.link.KNXNetworkLinkIP.sendRequestWait(KNXNetworkLinkIP.java:243) [bundleFile:?]
	at tuwien.auto.calimero.process.ProcessCommunicatorImpl.write(ProcessCommunicatorImpl.java:397) [bundleFile:?]
	at tuwien.auto.calimero.process.ProcessCommunicatorImpl.write(ProcessCommunicatorImpl.java:354) [bundleFile:?]
	at org.openhab.binding.knx.internal.client.AbstractKNXClient.sendToKNX(AbstractKNXClient.java:459) [bundleFile:?]
	at org.openhab.binding.knx.internal.client.AbstractKNXClient.writeToKNX(AbstractKNXClient.java:419) [bundleFile:?]
	at org.openhab.binding.knx.internal.handler.DeviceThingHandler.lambda$7(DeviceThingHandler.java:253) [bundleFile:?]
	at org.openhab.binding.knx.internal.handler.DeviceThingHandler.withKNXType(DeviceThingHandler.java:148) [bundleFile:?]
	at org.openhab.binding.knx.internal.handler.DeviceThingHandler.withKNXType(DeviceThingHandler.java:142) [bundleFile:?]
	at org.openhab.binding.knx.internal.handler.DeviceThingHandler.handleCommand(DeviceThingHandler.java:248) [bundleFile:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.openhab.core.internal.common.AbstractInvocationHandler.invokeDirect(AbstractInvocationHandler.java:154) [bundleFile:?]
	at org.openhab.core.internal.common.InvocationHandlerSync.invoke(InvocationHandlerSync.java:59) [bundleFile:?]
	at com.sun.proxy.$Proxy165.handleCommand(Unknown Source) [?:?]
	at org.openhab.core.thing.internal.profiles.ProfileCallbackImpl.handleCommand(ProfileCallbackImpl.java:80) [bundleFile:?]
	at org.openhab.core.thing.internal.profiles.SystemDefaultProfile.onCommandFromItem(SystemDefaultProfile.java:48) [bundleFile:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.openhab.core.internal.common.AbstractInvocationHandler.invokeDirect(AbstractInvocationHandler.java:154) [bundleFile:?]
	at org.openhab.core.internal.common.Invocation.call(Invocation.java:52) [bundleFile:?]
[... cut ...]
Bridge knx:ip:bridge [
    ipAddress="192.168.7.107",
    portNumber=3671,
    localIp="192.168.7.101",
    type="TUNNEL",
    readingPause=50,
    responseTimeout=10,
    readRetriesLimit=3,
    autoReconnectPeriod=30,     //optional, do not set <30 sec.
    localSourceAddr="0.1.242"
]
```

Same issue described here by @lewie : https://github.com/calimero-project/calimero-core/issues/33

This is drive me crazy...

If there is something I can do to help I'm available

Thank you for your help in advance...

bmalinowsky, 2021-01-22:
Why do you set `localSourceAddr`? Unless you have a reason to, don't do that for tunneling.

Anyway, from your log I don't see that a confirmation was sent. 
What you can try is the following: download Wireshark, select the network interface for `192.168.7.101`, then, in the text control "Apply a display filter" insert "kip" and see if a confirmation is received for the sequence you have shown above (or post the corresponding wshark log here). Depending on whether there actually is a confirmation, one could proceed.

villaRob, 2021-01-22:
Thanks @bmalinowsky, I already tried with wireshark, the missing confirmation depended on a wrong cEMI frame sent by my KNX-IP interface (WEINZIERL KNX IP BAOS 771). This wrong packet was discarded from calimero library I guess

I wrote to the Weinzierl and they reflashed device with a new firmware and resend back to me. So, seems that root cause of those problems was a bug in that old firmware.
Now I have two instances of openhab (1.8 and 3) running without problem without localSourceAddr set as you seggested.

Thanks!

bmalinowsky, 2021-03-11:
@PBrou did you fix the problem with your server? Does this issue still occur?

bmalinowsky, 2021-05-13:
Closing due to inactivity.

closed 2021-05-13 by bmalinowsky

PBrou, 2021-05-21:
I must admit I took the easy way out: I bought a Gira IP router instead.
No further testing was done on mentioned setup.
Greetz
Peter