calimero-project/calimero-core #89: Troubles using BAOS devices

petero-dk, 2019-10-04:
I have tried to use the latest Calimero with a kberry devices (on openhab) though openhab does not support the latest Calimero because it is based on Java 11 I have managed to get around that by crosscompiling it back to Java 8. That seems to work fine. It runs.

(I have also manually changed the source of Calimero to enable ft12 cemi and the baos specific command for mode switching)

However when I actually try to connect to the kberry I get lots of errors:

The first is:
```
Establishing connection to KNX bus through FT1.2 on serial port /dev/ttyAMA0.
tuwien.auto.calimero.KNXIllegalArgumentException: unknown L-Data message code f5
	at tuwien.auto.calimero.cemi.CEMILData.<init>(Unknown Source) ~[237:org.openhab.binding.knx:2.5.0.201910042000]
	at tuwien.auto.calimero.cemi.CEMIFactory.fromEmi(Unknown Source) ~[237:org.openhab.binding.knx:2.5.0.201910042000]
	at tuwien.auto.calimero.link.AbstractLink.onReceive(Unknown Source) ~[237:org.openhab.binding.knx:2.5.0.201910042000]
	at tuwien.auto.calimero.link.AbstractLink$LinkNotifier.frameReceived(Unknown Source) [237:org.openhab.binding.knx:2.5.0.201910042000]
	at tuwien.auto.calimero.serial.FT12Connection$Receiver.lambda$fireFrameReceived$0(Unknown Source) [237:org.openhab.binding.knx:2.5.0.201910042000]
	at tuwien.auto.calimero.serial.-$$Lambda$FT12Connection$Receiver$UgS9Su0XQrFi-m1ZRhp7KRwPs3M.accept(Unknown Source) [237:org.openhab.binding.knx:2.5.0.201910042000]
	at tuwien.auto.calimero.internal.EventListeners.fire(Unknown Source) [237:org.openhab.binding.knx:2.5.0.201910042000]
	at tuwien.auto.calimero.serial.FT12Connection$Receiver.fireFrameReceived(Unknown Source) [237:org.openhab.binding.knx:2.5.0.201910042000]
	at tuwien.auto.calimero.serial.FT12Connection$Receiver.readFrame(Unknown Source) [237:org.openhab.binding.knx:2.5.0.201910042000]
	at tuwien.auto.calimero.serial.FT12Connection$Receiver.run(Unknown Source) [237:org.openhab.binding.knx:2.5.0.201910042000]
```

Then the rest of my log file is littered with stuff like this
```
2019-10-04 23:04:15.908 [INFO ] [o.process.communication /dev/ttyAMA0] - timeout waiting for group read response from 2/2/7
2019-10-04 23:04:15.911 [WARN ] [nx.internal.client.AbstractKNXClient] - Giving up reading datapoint 2/2/7, the number of maximum retries (3) is reached.
2019-10-04 23:04:15.964 [DEBUG] [calimero.link./dev/ttyAMA0          ] - send message to 2/7/6, wait for ack
2019-10-04 23:04:16.058 [DEBUG] [calimero.link./dev/ttyAMA0          ] - confirmation of 15.15.255
2019-10-04 23:04:16.128 [DEBUG] [calimero.link./dev/ttyAMA0          ] - indication 11.12.224->1.1.171 L_Data.ind, system priority hop count 1, tpdu 06 01 00 40 00 00 00 00
2019-10-04 23:04:25.998 [INFO ] [o.process.communication /dev/ttyAMA0] - timeout waiting for group read response from 2/7/6
2019-10-04 23:04:26.001 [WARN ] [nx.internal.client.AbstractKNXClient] - Giving up reading datapoint 2/7/6, the number of maximum retries (3) is reached.
2019-10-04 23:04:26.055 [DEBUG] [calimero.link./dev/ttyAMA0          ] - send message to 2/4/9, wait for ack
2019-10-04 23:04:26.148 [DEBUG] [calimero.link./dev/ttyAMA0          ] - confirmation of 15.15.255
2019-10-04 23:04:26.228 [WARN ] [Data-unit builder                   ] - unknown APCI service code 0x102
2019-10-04 23:04:26.228 [DEBUG] [calimero.link./dev/ttyAMA0          ] - indication 11.12.224->1.1.141 L_Data.ind, system priority hop count 1, tpdu 09 02 00 40 02
2019-10-04 23:04:36.088 [INFO ] [o.process.communication /dev/ttyAMA0] - timeout waiting for group read response from 2/4/9
2019-10-04 23:04:36.091 [WARN ] [nx.internal.client.AbstractKNXClient] - Giving up reading datapoint 2/4/9, the number of maximum retries (3) is reached.
2019-10-04 23:04:36.145 [DEBUG] [calimero.link./dev/ttyAMA0          ] - send message to 2/7/7, wait for ack
2019-10-04 23:04:36.238 [DEBUG] [calimero.link./dev/ttyAMA0          ] - confirmation of 15.15.255
2019-10-04 23:04:36.308 [DEBUG] [calimero.link./dev/ttyAMA0          ] - indication 11.12.224->1.1.171 L_Data.ind, system priority hop count 1, tpdu 07 01 00 40 00 00 00 00
```

Which means I get nothing from the KNX network, however I can SEND to the knx network. That works just fine.

What would be a good way of proceeding with debugging here? Any tips?

petero-dk, 2019-10-04:
Btw. those indications are temperature readings from Vimar devices. My old setup used knxd which did actually work fine, it was not terribly stabile which is why I am pursuing using Calimero on the serial device directly instead of having knxd running also.

calimero-project, 2019-10-04:
The message code `f5` is a device mgmt message.

From what I see, the ft12 connection provides the frame as bytes, and the wrong factory is called (for EMI2, not the cEMI one). I missed that part in the patch somehow.

You can try in `AbstractLink.java::onReceive` (untested, replace the last line with the following two):

```
final byte[] bytes = e.getFrameBytes();
return f != null ? f : cEMI ? CEMIFactory.create(bytes,  0, bytes.length) : CEMIFactory.fromEmi(bytes);
```

(Which should create the correct cEMI devmgmt object)

petero-dk, 2019-10-04:
WUHUUUU

petero-dk, 2019-10-04:
I ask because once those two are enabled then I can actually create an OpenHab KNX Addon build with support for FT12 with Cemi and Baos :-)

calimero-project, 2019-10-05:
The fix will be there short-term with the next few commits ;)

For your second point, I created a new issue because the ObjectServer protocol is actually lower-level protocol agnostic, and also to be easier to follow for interested readers.

Please add any discussion points, requests, your usage, etc. there. This helps because I personally do not use any BAOS features.

closed 2020-06-16 by calimero-project