calimero-project/calimero-core #90: Support ObjectServer protocol of BAOS devices

calimero-project, 2019-10-05:
Evaluate approaches to enable ObjectServer protocol on top of communication protocols existing in Calimero, i.e., FT1.2, KNXnet/IP, USB, and RF. 

The initial focus would be FT1.2 (see also #89).

Implementation choices for network links:
1. One BCU switch to the desired mode (if necessary) during link creation, vs.
2. Dynamic switching based on request (link layer or ObjectServer)

The second choice is for completeness and not how links work today, but interesting in case upper layer datapoint/property services are serviced over a BAOS capable network link.

Others:
* Implications of existing BAOS group address filter for process communication

petero-dk, 2019-10-08:
My first idea was to put a configuration item in the constructors. However reading a little about the devices. It should be possible to detect it automatically but issuing commands and detecting the response?

I cannot figure out if that would be a viable way, however according to the docs of the baos device I can issue a serial number command (Page 41 in the following) that will return among other things (Hardware type/Hardware version ) and the version of the object server firmware:
https://weinzierl.de/images/download/development/830/KnxBAOS_Users_Guide.pdf

The Baos devices also has more information here:
https://www.weinzierl.de/images/download/development/830/KnxBAOS_Protocol_v2.pdf

Unfortunately I am not completely into the KNX protocol so I cant be more specific. If there is something I can research, please let me know.

bmalinowsky, 2019-10-09:
@petero-dk I created branch [feat/baos](https://github.com/calimero-project/calimero-core/tree/feat/baos) to test baos support in its various ways. 

I noticed one of my usb interfaces has baos support, so that what the current unit tests are based on ([here](https://github.com/calimero-project/calimero-core/blob/feat/baos/test/tuwien/auto/calimero/link/KNXNetworkLinkUsbTest.java)). Currently, I test reading baos properties, parameters, datapoints, and timers (which the usb interface does not support). 

So, ideally, one could start from there. Note that no baos events are currently forwarded to any link listeners, everything is done within the link. All work in progress.

petero-dk, 2020-01-03:
I have been running my own build of calimero for a couple of months with baos enabled and it was perfect :-) 

I have just updated to the latest source and dropped that in. That seems to work also.

bmalinowsky, 2020-06-16:
@petero-dk basic baos support is now in master.

closed 2020-06-16 by bmalinowsky