calimero-project/calimero-core #92: Maven artifacts are not deployed on maven central

FSeidinger, 2020-02-27:
Dear Calimero Team,

I tried to use your artifacts (calimero-core, calimero-device, and calimero-server) via maven central repository.

Whereas your parent is available for latest version 2.4, calimero-core is only available in version 2.4-rc2, calimero-device is also only available in version 2.4-rc2 and calimero-server is available in version 2.3.

This is in contrast to the released versions which are all 2.4 on github.

I could download the artifacts from github and install them locally, but this is not the intended way with resolving maven dependencies.

So please release the missing versions on maven central.

Kind regards,

Frank

calimero-project, 2020-02-27:
Version 2.4 can be downloaded from jcenter, which is also a maven repo. I didn't look into syncing it to maven central.

calimero-project, 2020-02-27:
I found a manual way to trigger the sync, let's see if they also show up on central.

FSeidinger, 2020-02-28:
That did the trick! Many thanks. I will cose the issue then.

closed 2020-02-28 by FSeidinger