calimero-project/calimero-core #93: No possibility to specify "No workday" in DPT19.001 String representation

NDahlem, 2020-04-02:
When a value is parsed in DPTXlatorDateTime.java there is only the possibility to specifiy "workday" in the string, resulting in WD=1 and NO_WD=0, omitting "workday" results in NO_WD=1, so there is no possibility to specify a bank day (no working day) (WD=0 and NO_WD=0).

I would suggest to expand the parser to either parse "no workday", collides with "no sync" and might make the implementation more complex, or parse "holiday" or "bankday".

bmalinowsky, 2020-04-05:
Thank you for reporting it! I will extend the parsing to accommodate for that field. 
I checked the spec, but I'm actually not familiar with "bank day"; do you know if this is common usage? I looked up wiki and ETS, they both use "bank holiday" (or no working day).

NDahlem, 2020-04-05:
I suspect that "bank holiday" as used in the UK is the intended meaning, as in holiday, not a working day. One could simply use "holiday" for parsing, or maybe "no workday" but the "no" is also used in "no sync"

closed 2020-04-06 by bmalinowsky

bmalinowsky, 2020-04-06:
Parsing now supports 'holiday', also the string value returned by the translator now uses 'holiday'. I think this output is clearer than using a negation like 'no workday'.

commit e0568831 (2020-04-06): Parse 'holiday' date/time working day information (closes #93)