calimero-project/calimero-core #97: New official release version?

petero-dk, 2021-01-17:
As openHAB 3 is not released and is running Java 11, it might be time for me to submit my changes to that project. Some of the changes are dependent on the current master of calimero (Thank you so much for helping getting that to work) 

The openhab project does however take issue with not running official versions and as far as I can see there are no official releases here on GitHub after 2.4?

Is there a timeline to release 2.5 or 3.0 or something else for the Calimero projects?

petero-dk, 2021-01-18:
More specifically what would be lovely is a new version released to https://bintray.com/bintray/jcenter

calimero-project, 2021-01-22:
The upcoming release is basically finished feature-wise. (There was the idea of doing a milestone release last summer for all the knx secure stuff and start rc in autumn, but since a lot of secure hw got delayed anyway, I didn't mind. So there's somewhat of a gap.) Stay tuned. All uploads will go to jcenter first.

bmalinowsky, 2021-03-11:
Update: since jcenter is dying, releases won't go there anymore. The M1 release is on maven only.

calimero-project, 2021-11-26:
v2.5 is the latest release.

closed 2021-11-26 by calimero-project