calimero-project/calimero-core #100: Does not work with tpuart ic

BbIKTOP, 2021-06-05:
Tried to make it work with tpuart, like ncn5120 with no success. Looks like it has no IC init/control at all. Is it possible to add it somehow? Ty!

bmalinowsky, 2021-06-06:
Can you clarify what did not work? The tpuart connection was implemented for the siemens ic. The ON semi service primitives are quite similar but might differ in details, I don't know.

The initialization is rather the same, i.e., sending a reset.req to the device.

BbIKTOP, 2021-06-06:
I simply bought the first available chip, made a "reference schematic" board with chip configured for uart, compiled sources and ran the test monitoring example on the raspberry pi. Got no output at all, it's completely silent. The calimero architecture is a bit difficult to understand, but as I see in the datasheet, this chip does not monitor the tp until inited. I wrote a simple program that sends 0x01 to the chip ("reset service" command), ran it on the same raps, and immediately observed that monitoring has been started (i've got 0x03 confirmation response, then all telegrams that I send on the same tp). I suppose, I could add the support, but have no idea where to start. Spent a few hours trying to understand calimero's architecture design, but didn't succeed much yet. Could you advise me some readings, like, documentation that could help?
Btw, what rxtx do you use? The one that was downloaded by maven using your pom.xml, does not work, throws some exception about blocking. I had to replace it with the one I found on some openhab's forums, and either your and mine monitoring started to work with it. Although your example produces no output.

BbIKTOP, 2021-06-06:
Well, I suppose, I could simplify my question. Is there a place in the code, where I could define IC commands, like "reset", "state switch" "send/receive frame" etc?

bmalinowsky, 2021-06-06:
All the tpuart stuff is defined in a single file [here](https://github.com/calimero-project/calimero-core/blob/master/src/tuwien/auto/calimero/serial/TpuartConnection.java).

The first thing the tpuart connection does is send a [reset.req](https://github.com/calimero-project/calimero-core/blob/ee09f14792b2e90d5225c1642a3f4187df9a078e/src/tuwien/auto/calimero/serial/TpuartConnection.java#L192), i.e., 0x01, and then wait for a reset.ind (0x03) in the [receiver loop](https://github.com/calimero-project/calimero-core/blob/ee09f14792b2e90d5225c1642a3f4187df9a078e/src/tuwien/auto/calimero/serial/TpuartConnection.java#L572).

And if you have trace level enabled, you should see something like this:
```
23:26:19:482 DEBUG calimero.serial.tpuart:/dev/tty.usbmodem14201 - using rxtx library for serial port access
23:26:19:568 INFO calimero.serial.tpuart:/dev/tty.usbmodem14201 - open rxtx (RXTX-2.2pre2) serial port connection for /dev/tty.usbmodem14201
23:26:19:582 DEBUG calimero.serial.tpuart:/dev/tty.usbmodem14201 - setup serial port: baudrate 19200, even parity, 8 databits, 1 stopbits, flow control 0
23:26:19:583 DEBUG calimero.serial.tpuart:/dev/tty.usbmodem14201 - reset TP-UART controller
23:26:19:583 TRACE calimero.serial.tpuart:/dev/tty.usbmodem14201 - wait for initial TP-UART state
23:26:19:590 TRACE calimero.serial.tpuart:/dev/tty.usbmodem14201 - drained rx queue (1 bytes)
23:26:19:590 DEBUG calimero.serial.tpuart:/dev/tty.usbmodem14201 - TP-UART reset.ind
23:26:19:590 TRACE calimero.serial.tpuart:/dev/tty.usbmodem14201 - loop time = 409 us
23:26:19:597 TRACE calimero.serial.tpuart:/dev/tty.usbmodem14201 - TP-UART status: Temp. OK
23:26:19:597 TRACE calimero.serial.tpuart:/dev/tty.usbmodem14201 - loop time = 1506 us
23:26:19:609 DEBUG calimero.serial.tpuart:/dev/tty.usbmodem14201 - activate TP-UART busmonitor
23:26:19:610 INFO calimero.link.tpuart:/dev/tty.usbmodem14201 - enable decoding of raw frames
Connection established
```

I used this example program in the introduction repo: [tpuart monitor](https://github.com/calimero-project/introduction/blob/master/src/main/java/CreateTpuartMonitor.java)

So this should indicate at which step there is the issue when communicating with the nc51xx chips.

The rxtx adapter is in this repo: [adapter](https://github.com/calimero-project/calimero-rxtx)
It adds 'com.neuronrobotics:nrjavaserial:5.2.1' as dependency.
(If you're more familiar with C(++) you can also use [the jni lib](https://github.com/calimero-project/serial-native) instead. Calimero will look for this lib first in the java path.)

BbIKTOP, 2021-06-07:
Yes, tpuart monitor is is the example I've tried. It threw some blocking exception and I replaced nrjavaserial with another one I've found. I thought there's something wrong with the communications at first, so I've made a simple program that just sends 0x01, then waits for 0x03, then prints everything it receives. It works fine with the same nrjavaserial. Will try it tomorrow and let you know, thank you! I tested it with the 2.4 release, is it correct? Shall I use 2.5 probably?

bmalinowsky, 2021-06-07:
Yes, better to use 2.5 because any fix etc. that is necessary will be in there (not in 2.4).
Anyway, I tried now with the 2.4 version and I also don't get any error. So from that I can't say where your problem with the blocking exception comes from.

nremennik, 2021-06-07:
We're continuing to investigate this.
Using native serial:
```
This example establishes a KNX monitor connection using TP-UART on port '/dev/ttyAMA0'
2021-06-07 15:41:49:139 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - open Calimero native serial port connection (serialcom) for /dev/ttyAMA0
2021-06-07 15:41:52:458 +0300 [main] WARN calimero.serial.tpuart:/dev/ttyAMA0 - no rxtx library adapter found
Exception in thread "main" tuwien.auto.calimero.KNXException: failed to open serial port /dev/ttyAMA0
	at tuwien.auto.calimero.serial.LibraryAdapter.open(LibraryAdapter.java:160)
	at tuwien.auto.calimero.serial.TpuartConnection.<init>(TpuartConnection.java:176)
	at tuwien.auto.calimero.link.KNXNetworkMonitorTpuart.<init>(KNXNetworkMonitorTpuart.java:71)
	at CreateTpuartMonitor.main(CreateTpuartMonitor.java:32)
Caused by: java.io.IOException: Success
	at tuwien.auto.calimero.serial.SerialComAdapter.open(Native Method)
	at tuwien.auto.calimero.serial.SerialComAdapter.<init>(SerialComAdapter.java:166)
	at tuwien.auto.calimero.serial.LibraryAdapter.open(LibraryAdapter.java:123)
	... 3 more
```

nremennik, 2021-06-07:
Although our user is in the dialout group, we've got better results running it with sudo:
```
This example establishes a KNX monitor connection using TP-UART on port '/dev/ttyAMA0'
2021-06-07 15:51:53:630 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - open Calimero native serial port connection (serialcom) for /dev/ttyAMA0
2021-06-07 15:51:53:982 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - setup serial port: baudrate 19200, even parity, 8 databits, 1 stopbits, timeouts: read 0 read total 0 constant 100 write total 0 write constant 0
2021-06-07 15:51:53:992 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - reset TP-UART controller
Exception in thread "main" tuwien.auto.calimero.serial.KNXPortClosedException: timeout waiting for initial TP-UART state
	at tuwien.auto.calimero.serial.TpuartConnection.<init>(TpuartConnection.java:194)
	at tuwien.auto.calimero.link.KNXNetworkMonitorTpuart.<init>(KNXNetworkMonitorTpuart.java:71)
	at CreateTpuartMonitor.main(CreateTpuartMonitor.java:32)
```

nremennik, 2021-06-07:
Also, with nrjavaserial. using the same lib with this test:
```Java
import gnu.io.NRSerialPort;
import gnu.io.SerialPort;

import java.io.DataInputStream;
import java.io.DataOutputStream;

public class KnxUart
{

    public static void main(String[] args)
    {
        String port = "/dev/ttyAMA0";
        for (String s : NRSerialPort.getAvailableSerialPorts())
        {
            System.out.println("Availible port: " + s);
            port = s;
        }

        int baudRate = 19200;
        NRSerialPort serial = new NRSerialPort(port, baudRate);
        serial.setDataBits(8);
        serial.setParity(SerialPort.PARITY_EVEN);
        serial.setStopBits(SerialPort.STOPBITS_1);
        serial.connect();

        DataInputStream ins = new DataInputStream(serial.getInputStream());
        DataOutputStream outs = new DataOutputStream(serial.getOutputStream());
        try
        {
            outs.write(1);
            int counter = 0;
            while (!Thread.interrupted())
            {// read all bytes
                if (ins.available() > 0)
                {
                    byte b = (byte) ins.read();
                    //outs.write((byte)b);
                    System.out.printf("%2X ", b);

                    if (++counter > 20)
                    {
                        System.out.println();
                        counter = 0;
                    }
                }
                Thread.sleep(5);
            }
        }
        catch (Exception ex)
        {
            ex.printStackTrace();
        }
        serial.disconnect();
    }
}
```

Output:

```
Availible port: /dev/ttyS0
 3 BC 47 16 1A  3 E2  0 80  0 69 9C 47 16 1A  3 E2  0 80  0 49 
9C 47 16 1A  3 E2  0 80  0 49 9C 47 16 1A  3 E2  0 80  0 49 BC 
47 16 1A  3 E2  0 80 FF 96 9C 47 16 1A  3 E2  0 80 FF B6 9C 47 
16 1A  3 E2  0 80 FF B6 9C 47 16 1A  3 E2  0 80 FF B6 
```

Calimero monitor example output:
```
This example establishes a KNX monitor connection using TP-UART on port '/dev/ttyAMA0'
2021-06-07 16:06:46:721 +0300 [main] INFO calimero.serial - no serialcom in java.library.path: [/usr/java/packages/lib, /usr/lib/arm-linux-gnueabihf/jni, /lib/arm-linux-gnueabihf, /usr/lib/arm-linux-gnueabihf, /usr/lib/jni, /lib, /usr/lib]
2021-06-07 16:06:46:802 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - using rxtx library for serial port access
2021-06-07 16:06:47:022 +0300 [main] INFO calimero.serial.tpuart:/dev/ttyAMA0 - open rxtx (RXTX-2.2pre2) serial port connection for /dev/ttyAMA0
2021-06-07 16:06:47:073 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - setup serial port: baudrate 19200, even parity, 8 databits, 1 stopbits, flow control 0
2021-06-07 16:06:47:087 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - reset TP-UART controller
Exception in thread "main" tuwien.auto.calimero.serial.KNXPortClosedException: timeout waiting for initial TP-UART state
	at tuwien.auto.calimero.serial.TpuartConnection.<init>(TpuartConnection.java:194)
	at tuwien.auto.calimero.link.KNXNetworkMonitorTpuart.<init>(KNXNetworkMonitorTpuart.java:71)
	at CreateTpuartMonitor.main(CreateTpuartMonitor.java:32)
```

nremennik, 2021-06-07:
That was using 2.5. using 2.4 with nr:
```
This example establishes a KNX monitor connection using TP-UART on port '/dev/ttyAMA0'
2021-06-07 16:22:19:005 +0300 [main] INFO calimero.serial - no serialcom in java.library.path: [/usr/java/packages/lib, /usr/lib/arm-linux-gnueabihf/jni, /lib/arm-linux-gnueabihf, /usr/lib/arm-linux-gnueabihf, /usr/lib/jni, /lib, /usr/lib]
2021-06-07 16:22:19:090 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - using rxtx library for serial port access
2021-06-07 16:22:19:211 +0300 [main] INFO calimero.serial.tpuart:/dev/ttyAMA0 - open rxtx (RXTX-2.2pre2) serial port connection for /dev/ttyAMA0
2021-06-07 16:22:19:254 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - setup serial port: baudrate 19200, even parity, 8 databits, 1 stopbits, flow control 0
2021-06-07 16:22:19:266 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - reset TP-UART controller
2021-06-07 16:22:19:282 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - activate TP-UART busmonitor
2021-06-07 16:22:19:313 +0300 [main] INFO calimero.link.tpuart:/dev/ttyAMA0 - enable decoding of raw frames
Connection established
Adding listener
2021-06-07 16:22:19:518 +0300 [Calimero TP-UART receiver] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - reset TP-UART controller
2021-06-07 16:22:19:534 +0300 [Calimero TP-UART receiver] INFO calimero.link.tpuart:/dev/ttyAMA0 - monitor closed
receiver communication failure : connection closed: 3
```

nremennik, 2021-06-07:
With native serialcomm and 2.4 (almost the same):
``
This example establishes a KNX monitor connection using TP-UART on port '/dev/ttyAMA0'
2021-06-07 16:27:26:252 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - open Calimero native serial port connection (serialcom) for /dev/ttyAMA0
2021-06-07 16:27:26:602 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - setup serial port: baudrate 19200, even parity, 8 databits, 1 stopbits, timeouts: read 0 read total 0 constant 200 write total 0 write constant 0
2021-06-07 16:27:26:607 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - reset TP-UART controller
2021-06-07 16:27:26:620 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - activate TP-UART busmonitor
2021-06-07 16:27:26:649 +0300 [main] INFO calimero.link.tpuart:/dev/ttyAMA0 - enable decoding of raw frames
Connection established
Adding listener
2021-06-07 16:27:26:859 +0300 [Calimero TP-UART receiver] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - reset TP-UART controller
2021-06-07 16:27:26:872 +0300 [Calimero TP-UART receiver] INFO calimero.link.tpuart:/dev/ttyAMA0 - monitor closed
receiver communication failure : connection closed: 3
``

bmalinowsky, 2021-06-07:
Regarding sudo: /dev/ttyAMA0 is in the dialout group, right? Did you freshly add the group, sometimes re-login / reboot is necessary to take effect. Otherwise, where did you put the serialcom lib? Is this accessible without root privileges? It should always work if put into the cwd.

From the logs (can you please do TRACE logs): the reset.req/.ind actually do work, but then there is an immediate close notification.

Also, in your KnxUart example, you assign `/dev/ttyS0` as port, in the other examples you use AMA0? Which one is the correct one?

nremennik, 2021-06-07:
Now it behaves differently. It starts but does not receive no telegrams. Have no idea why, we recompiled serialcom and tested with 2.4.

About sudo - we have localized this problem, it was related to bt. Raps' linups did start hciattach after the first run, now we have bt removed and the problem has gone.
serialcom is in the LD_LIBRARY_PATH directory /home/knx/dlib/:

```
$ ls -l /home/knx/dlib/libserialcom.so
-rwxr-xr-x 1 knx knx 19532 Jun  7 17:11 /home/knx/dlib/libserialcom.so

```

In our test code we use ttyAMA0 and it works fine:

```Java
public class KnxUart
{

    public static void main(String[] args)
    {
        String port = "/dev/ttyAMA0";
```
As well as in the CreateTpuartMonitor:

 Surely, it is in the dialout group as our test works fine:
```
$ ls -l /dev/ttyAMA0
crw-rw---- 1 root dialout 204, 64 Jun  7 17:39 /dev/ttyAMA0
```
It has been in this group by default, we did some reboots, of course, with no success (result is the same)

Output with "trace" (rxtx version):
```
This example establishes a KNX monitor connection using TP-UART on port '/dev/ttyAMA0'
2021-06-07 17:47:58:169 +0300 [main] TRACE calimero.serial - check Java library path /usr/java/packages/lib:/usr/lib/arm-linux-gnueabihf/jni:/lib/arm-linux-gnueabihf:/usr/lib/arm-linux-gnueabihf:/usr/lib/jni:/lib:/usr/lib
2021-06-07 17:47:58:177 +0300 [main] INFO calimero.serial - no serialcom in java.library.path: [/usr/java/packages/lib, /usr/lib/arm-linux-gnueabihf/jni, /lib/arm-linux-gnueabihf, /usr/lib/arm-linux-gnueabihf, /usr/lib/jni, /lib, /usr/lib]
2021-06-07 17:47:58:213 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - using rxtx library for serial port access
2021-06-07 17:47:58:358 +0300 [main] INFO calimero.serial.tpuart:/dev/ttyAMA0 - open rxtx (RXTX-2.2pre2) serial port connection for /dev/ttyAMA0
2021-06-07 17:47:58:398 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - setup serial port: baudrate 19200, even parity, 8 databits, 1 stopbits, flow control 0
2021-06-07 17:47:58:404 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - reset TP-UART controller
2021-06-07 17:47:58:417 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - activate TP-UART busmonitor
2021-06-07 17:47:58:446 +0300 [main] INFO calimero.link.tpuart:/dev/ttyAMA0 - enable decoding of raw frames
Connection established
Adding listener
2021-06-07 17:47:58:657 +0300 [Calimero TP-UART receiver] TRACE calimero.serial.tpuart:/dev/ttyAMA0 - drained rx queue (0 bytes)
2021-06-07 17:49:38:450 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - reset TP-UART controller
```

Output with "trace" (serialcomm version):
```
This example establishes a KNX monitor connection using TP-UART on port '/dev/ttyAMA0'
2021-06-07 17:57:27:726 +0300 [main] TRACE calimero.serial - check Java library path /home/knx/dlib/
2021-06-07 17:57:27:747 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - open Calimero native serial port connection (serialcom) for /dev/ttyAMA0
2021-06-07 17:57:28:095 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - setup serial port: baudrate 19200, even parity, 8 databits, 1 stopbits, timeouts: read 0 read total 0 constant 200 write total 0 write constant 0
2021-06-07 17:57:28:101 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - reset TP-UART controller
2021-06-07 17:57:28:135 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - activate TP-UART busmonitor
2021-06-07 17:57:28:190 +0300 [main] INFO calimero.link.tpuart:/dev/ttyAMA0 - enable decoding of raw frames
Connection established
Adding listener
2021-06-07 17:57:28:352 +0300 [Calimero TP-UART receiver] TRACE calimero.serial.tpuart:/dev/ttyAMA0 - drained rx queue (0 bytes)
2021-06-07 17:59:08:195 +0300 [main] DEBUG calimero.serial.tpuart:/dev/ttyAMA0 - reset TP-UART controller
user request : connection closed: 0
2021-06-07 17:59:08:392 +0300 [main] INFO calimero.link.tpuart:/dev/ttyAMA0 - monitor closed
```

No errors, but still no telegrams seen.

bmalinowsky, 2021-06-07:
I asked about the tty device, because in your test program you assign the found serial port `/dev/ttyS0` to `port`. At least from the code you posted.

nremennik, 2021-06-07:
Ah, I see! You mean this: ``Availible port: /dev/ttyS0``? This is nrjava's output, it shows all available ports. Looks like output was accidentally cut off, sorry. In the source it is
```Java
public static void main(String[] args)
    {
        String port = "/dev/ttyAMA0";
```
Full output is:
```
$ java -jar knxcom.jar
Availible port: /dev/ttyAMA0
Availible port: /dev/ttyS0
 3 BC 47 16  8  0 E1  0 80 7B 9C 47 16  8  0 E1  0 80 5B 9C 47
16  8  0 E1  0 80 5B 9C 47 16  8  0 E1  0 80 5B BC 47 16  8  0
E1  0 81 7A 9C 47 16  8  0 E1  0 81 5A 9C 47 16  8  0 E1  0 81
5A 9C 47 16  8  0 E1  0 81 5A BC 47 16 2A  7 E2  0 80  0 5D 9C
47 16 2A  7 E2  0 80  0 7D 9C 47 16 2A  7 E2  0 80  0 7D 9C 47
16 2A  7 E2  0 80  0 7D
```

nremennik, 2021-06-07:
Well, now I see. Really ))) Will check and report back, thank you!

nremennik, 2021-06-07:
Yes, sorry about that, it was our fault! Strange, because all docs refers to raps uart as AMA0, but S0 is working. Please close this. Thank you!

BbIKTOP, 2021-06-07:
Can confirm, it was my mistake, sorry and thank you! Btw, any docs about how to send telegram somewhere? ;)

closed 2021-06-08 by bmalinowsky

BbIKTOP, 2021-06-09:
Could I use monitor and KNXNetworkLinkTpuart simultaneously? Is there any readings or examples? Thank you!

bmalinowsky, 2021-06-09:
No you can't, a device only supports one layer at a time (either link layer or bus monitor layer).

BbIKTOP, 2021-06-09:
Yes, I know it from the datasheet, but I supposed it is possible to enter monitoring mode to observe bus telegrams, and if there's a need to send a telegram, to configure device's address, send telegram, get answers if any, then to put the device in the monitoring state againn? Or any other possible scenario?

bmalinowsky, 2021-06-09:
You can also receive all the relevant telegrams if you're not in monitoring mode, I pointed you to the group monitor example for process communication which does exactly that. The specific "bus monitor" layer a device offers is interesting mainly for diagnostic purposes, in that it is a passive listener. If you actually want to switch between those two modes, you close the monitor/link and open the other (i.e., link/monitor), which will initialize the device to do exactly that.

BbIKTOP, 2021-06-09:
Ok, will try, thank you!