calimero-project/calimero-core #104: Mangamementclient no basic restart anymore with v2.5?

Darthyson, 2021-11-30:
Hi,
with calimero-core v2.5 calling ManagementClientImpl.restart(dst, 0, 255) results in:
```
tuwien.auto.calimero.KNXIllegalArgumentException: unsupported erase code 0
	at tuwien.auto.calimero.mgmt.ManagementClient$EraseCode.of(ManagementClient.java:492)
	at tuwien.auto.calimero.mgmt.ManagementClient.restart(ManagementClient.java:565)
	...
```

Is this behavior intended? I think basic restart is still supported by the KNX spec.
The lowest EraseCode i can find in ManagementClient.java is _ConfirmedRestart_ which results in 1.

bmalinowsky, 2021-11-30:
The method you are calling is restarting a device including a master reset, not a basic restart. A basic restart would be `::restart(dst)`.

Darthyson, 2021-11-30:
Problem is on my site. I just read the spec, Btw. it was working with 2.4 :-). Thanks for the fast response.

closed 2021-11-30 by Darthyson

bmalinowsky, 2021-11-30:
Thanks for the update. 
Yes, before 2.5 the method did no range check of the erase code before sending. But it shouldn't have worked because a remote endpoint should not restart itself with code 0 (ideally, maybe some do).