calimero-project/calimero-core #105: [Feature Request] "protected" for some fields and methods of ManagementClientImpl

Darthyson, 2021-12-15:
Hi,
it would be nice if some fields and methods of `ManagementClientImpl `were not private but protected. 
I know that i could implement the `ManagementClient interface` by myself, but this would be a 99% copy of `ManagementClientImpl`. My goal is to write a modified version of `writeMemory(...)`

For that I would need the following:

```
- protected final TransportLayer tl
- protected long send(...)
- protected byte[] sendWait(...)
```
At the moment I use reflections to achieve this, but I would like to avoid that.

bmalinowsky, 2021-12-16:
Hi,
yes, overriding those methods is currently not optimal. 

`tl.sendData` should actually be replaced with the standard `send` method, so `writeMemory` essentially requires `send` & `sendWait`.

If you need access to tl, you can also use the protected ctor. It's behavior is different when detaching (if you pass in the transport layer you own it, otherwise the mgmt client owns it)

Darthyson, 2021-12-23:
Hi,
i tried using the protected ctor like this:
```
public SB2ManagementClientImpl(KNXNetworkLink link)  
        throws KNXLinkClosedException {  
    tl = new TransportLayerImpl(link);  
    super(link, tl);  
    ...  
}
```
but this leads to the problem, that super(..) must be the first call and i don't want to create the tl "outside", actually i don't want to handle the tl at all, instead ManagementClientImpl should handle it for me.
I could avoid that by using `send` and `sendWait`, but they are private.

If you are interested, under [https://github.com/selfbus/software-arm-lib/.../SBManagementClientImpl.java](https://github.com/selfbus/software-arm-lib/blob/DEV-Bootloader/Bus-Updater/PC_updater_tool/source/src/main/java/org/selfbus/updater/SBManagementClientImpl.java)  you'll find my dirty workaround.

bmalinowsky, 2022-01-25:
Hi @Darthyson,
a few methods are now `protected`, so I think your use-case should work without reflection (there is no snapshot yet).

Darthyson, 2022-01-27:
Hi,
first tests were successful. Thanks a lot. That makes life much easier. Is there a timeline for a v2.6-snapshot? calimero-rxtx is already pointing to it.

bmalinowsky, 2022-01-27:
The latest snapshot now contains that commit.

Darthyson, 2022-01-28:
OK, i was missing the snapshot repository...
````
maven { url 'https://oss.sonatype.org/content/repositories/snapshots' }
````
Thanks again.

closed 2022-01-28 by Darthyson