calimero-project/calimero-core #107: Receiving a GroupValueResponse with DPT 1.001 fails due to missing data in the asdu

FSeidinger, 2022-02-18:
My setup is as follows:

- Calimero version 2.5
- Java OpenJDK 64-Bit Server VM Temurin-11.0.13+8
- KNX IP Interface in router mode

Communication is with DPT 1.001 on group addresses 9/1/15 for switching the actor and 9/1/18 for receiving the state from the actor.

I've setup a KNX client using the `ProcessCommunicator` and implementing the `ProcessListener` interface. Handling `GroupValueWrite` telegrams to 9/1/15 works. The client picks up the telegram and sends it via `ProcessEvent` to the listener.

After that I've tested the `GroupValueResponse`. I've emitted a GroupValueRead telegram to 9/1/18 and the actor answers with a GroupValueResponse to the bus. This client picks up the telegram and sends it via `ProcessEvent` to the listener. My listener cannot process the event, because the data in the asdu byte array is empty. Trying to parse the response with the DTPXlatorBoolean gives an error indicating that the data length is zero. I've logged the event and the `[]` indicates that it is coming with zero data to the listener.

```
Received group read response from device 1.0.1 for group address 9/1/18 - []
tuwien.auto.calimero.KNXIllegalArgumentException: data length 0 < KNX data type width 1
        at tuwien.auto.calimero.dptxlator.DPTXlatorBoolean.setData(DPTXlatorBoolean.java:303) ~[bundleFile:?]
        at tuwien.auto.calimero.dptxlator.DPTXlator.setData(DPTXlator.java:235) ~[bundleFile:?]
        at org.openhab.binding.knxng.core.domain.CommunicationObject.update(CommunicationObject.java:93) ~[bundleFile:?]
        at org.openhab.binding.knxng.core.application.EventReceiverService.delegate(EventReceiverService.java:108) ~[bundleFile:?]
        at org.openhab.binding.knxng.core.application.EventReceiverService.groupReadRequest(EventReceiverService.java:73) ~[bundleFile:?]
        at tuwien.auto.calimero.process.ProcessCommunicatorImpl$NLListener.lambda$fireGroupReadWrite$0(ProcessCommunicatorImpl.java:128) ~[bundleFile:?]
        at tuwien.auto.calimero.internal.EventListeners.fire(EventListeners.java:137) ~[bundleFile:?]
        at tuwien.auto.calimero.process.ProcessCommunicatorImpl$NLListener.fireGroupReadWrite(ProcessCommunicatorImpl.java:133) ~[bundleFile:?]
        at tuwien.auto.calimero.process.ProcessCommunicatorImpl$NLListener.indication(ProcessCommunicatorImpl.java:113) ~[bundleFile:?]
        at tuwien.auto.calimero.secure.SecureApplicationLayer.lambda$dispatchLinkEvent$7(SecureApplicationLayer.java:748) ~[bundleFile:?]
        at tuwien.auto.calimero.internal.EventListeners.fire(EventListeners.java:137) ~[bundleFile:?]
        at tuwien.auto.calimero.secure.SecureApplicationLayer.dispatchLinkEvent(SecureApplicationLayer.java:748) ~[bundleFile:?]
        at java.util.Optional.ifPresent(Optional.java:183) ~[?:?]
        at tuwien.auto.calimero.secure.SecureApplicationLayer$1.indication(SecureApplicationLayer.java:173) ~[bundleFile:?]
        at tuwien.auto.calimero.link.AbstractLink$LinkNotifier.lambda$frameReceived$0(AbstractLink.java:193) ~[bundleFile:?]
        at tuwien.auto.calimero.internal.EventListeners.fire(EventListeners.java:137) [bundleFile:?]
        at tuwien.auto.calimero.link.EventNotifier.fire(EventNotifier.java:169) [bundleFile:?]
        at tuwien.auto.calimero.link.EventNotifier.run(EventNotifier.java:88) [bundleFile:?]
```
Maybe this is due to special handling of data less than 6 bits? In the code this looks like a data length of 0 bytes is implied by that.

bmalinowsky, 2022-02-18:
Why is your stack trace showing a `EventReceiverService::groupReadRequest`? A read request does not carry any datapoint data.

FSeidinger, 2022-02-18:
Sorry, my mistake. My delegation was erroneous.

closed 2022-02-18 by FSeidinger