calimero-project/calimero-core #108: DPT 251.600 format should be consistent across platforms and locales

J-N-K, 2022-03-18:
Currently depending on platform and locale the same bytes `26 2B 31 00 00 0E` received from the bus result in different representation `14,9 16,9 19,2 - %` or `14.9 16.9 19.2 - %`. This results in unnecessary processing in code that uses the response value. The output should be the same everywhere.

calimero-project, 2022-03-18:
Using a different Locale will not result in the same output everywhere. Users usually select a different Locale for a reason, no?

J-N-K, 2022-03-18:
That's exactly my point: different locales will result in different outputs. Is this value really intended for displaying it to the user (in that case it makes sense to use a localized value) or is it intended for further processing by machines? If the latter, then a fixed format (with either `.` or `,` as decimal point) would be much preferred. In my application I split the string and use `Double.parseDouble` on the values.

bmalinowsky, 2022-10-21:
@J-N-K In the previous weeks I reviewed most of the translator implementations for some other reason, and parsing under-specified string output should not be the preferred way of obtaining required translator information *. Changing the output to the current Locale is in fact for the purpose to present a familiar representation to the user wrt the current OS/JVM settings.

If you like, we can discuss a static setter to overrule a specific Locale, or missing accessors, or a deterministic way to parse strings (or any other idea you have).

* I am aware that in older versions of Calimero certain accessors were missing which required parsing strings.

closed 2023-09-19 by calimero-project