calimero-project/calimero-core #110: Android project runtime FATAL EXCEPTION.

anand-010, 2022-04-19:
I've tried to implement calimero-core on an android project.
`implementation("com.github.calimero:calimero-core:2.5")`
But i've got this error every time application starts.
```
2022-04-18 21:22:12.293 25568-25568/com.example.knxmate E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.example.knxmate, PID: 25568
    java.lang.NoSuchMethodError: No static method networkInterfaces()Ljava/util/stream/Stream; in class Ljava/net/NetworkInterface; or its super classes (declaration of 'java.net.NetworkInterface' appears in /apex/com.android.art/javalib/core-oj.jar)
        at tuwien.auto.calimero.knxnetip.Net.onSameSubnet(Net.java:74)
        at tuwien.auto.calimero.knxnetip.Net$$ExternalSyntheticLambda1.apply(Unknown Source:2)
        at java.util.Optional.flatMap(Optional.java:237)
        at tuwien.auto.calimero.knxnetip.Net.matchRemoteEndpoint(Net.java:111)
        at tuwien.auto.calimero.knxnetip.ClientConnection.connect(ClientConnection.java:174)
        at tuwien.auto.calimero.knxnetip.KNXnetIPTunnel.<init>(KNXnetIPTunnel.java:171)
        at tuwien.auto.calimero.knxnetip.KNXnetIPTunnel.<init>(KNXnetIPTunnel.java:163)
        at tuwien.auto.calimero.link.KNXNetworkLinkIP.newConnection(KNXNetworkLinkIP.java:486)
        at tuwien.auto.calimero.link.KNXNetworkLinkIP.<init>(KNXNetworkLinkIP.java:267)
        at tuwien.auto.calimero.link.KNXNetworkLinkIP.newTunnelingLink(KNXNetworkLinkIP.java:150)
        at com.example.knxmate.MainActivity.onCreate(MainActivity.kt:28)
        at android.app.Activity.performCreate(Activity.java:8109)
        at android.app.Activity.performCreate(Activity.java:8083)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1309)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3493)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3692)
        at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:85)
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2142)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:236)
        at android.app.ActivityThread.main(ActivityThread.java:7864)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:620)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1011)
```
The error was occurred in the line.
```
        KNXNetworkLinkIP.newTunnelingLink(local, server, false, TPSettings())
```
```
val local = InetSocketAddress(0)
val server = InetSocketAddress("192.168.1.11", KNXnetIPConnection.DEFAULT_PORT)
```
What causes this error?.
How do we solve it?.

bmalinowsky, 2022-04-19:
Calimero v2.5 requires jdk 11. Your error shows that the program cannot find a method which was added in jdk 9.

anand-010, 2022-04-19:
[image]
I've tried different jdk's in android studio but non of them worked.
Here is my [activity.kt](https://jpst.it/2P3oZ)
Please guide me.

bmalinowsky, 2022-04-19:
Did you also set the corresponding minimum `sourceCompatibility` and `targetCompatibility` in the gradle file? AFAIR it defaulted to 7 for a long time. Something like

```
android {
  compileOptions {
    sourceCompatibility JavaVersion.VERSION_11
    targetCompatibility JavaVersion.VERSION_11
  }
}
```

anand-010, 2022-04-19:
yeh i've tried that too.
```
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_11
        targetCompatibility JavaVersion.VERSION_11
    }
    kotlinOptions {
        jvmTarget = '11'
    }
```
Same result do you have a working project  ?.
please send it as a zipped file.

bmalinowsky, 2022-04-19:
I don't have an Android-specific gradle project, just the gradle files you find in this repo for standard jdks. 
Anyway, I looked up the Android `NetworkInterface` doc [here](https://developer.android.com/reference/java/net/NetworkInterface), and it simply doesn't offer that method.

You might want to use Calimero v2.4, which only requires jdk 8. Android support is probably better.

closed 2022-05-01 by bmalinowsky