calimero-project/calimero-core #115: Throws Exception when EventListener catch one

bgrand-Lam, 2022-07-26:
When the EventListener catches an exception, it is not possible to know if the listener is removed. And at this point the KNX service can't work after that.

The proposal is to throw an exception when it appears. And this exception can be caught on the override method of ProcessListener or in a special method "detachListener".

If a another method to know if the listener is removed I would love to hear about it.

Calimero version : 2.5

bmalinowsky, 2022-07-27:
Why don't you catch your exceptions directly in the event listener?

bgrand-Lam, 2022-07-27:
Yes, after a few tries, it is now good. The problem is that the exception is a KNXRuntimeException and I have been caught KNXException. I thought the KNXException was the parent.

closed 2022-07-27 by bgrand-Lam