calimero-project/calimero-core #121: Calimero GUI and KNX Virtual

GCnl85, 2023-05-15:
Hello, I was trying to do some test connecting the Calimero GUI to KNX Virtual v2.5.

The tunneling connection is established, but then Calimero sends a configuration request message which seems to be ignored by the interface exposed by KNX Virtual, so the interface is disconected.

Did you already experienced something similar? Is it - maybe - a known issue?

Attached you find:
 - Wireshark capture
 - A screenshot of the GUI trace

Let me know if you need something else and thanks in advance!

[image]
[calimerogui_knxvirtual_capture.zip](https://github.com/calimero-project/calimero-core/files/11475949/calimerogui_knxvirtual_capture.zip)

bmalinowsky, 2023-05-15:
Hello,
the warning only concerns the management connection, the tunneling connection used for communication is not affected.
This can happen (I've seen it on some occasions), could be that KNX Virtual simple doesn't implement it.

GCnl85, 2023-07-31:
I close this issue for now, as I didnt hav any chance to further investigate on it. Thanks!