calimero-project/calimero-core #122: Data Secure - Read Group Request management

GCnl85, 2023-07-31:
Hello, 

I'm trying to do some test with a Java application I'm working on, implementing the build 2.5.1. 

My objective consists of sending a Read Group Request and receive the answer for a secured group, through a tunneler device.
So I've prepared - with ETS 6 - the test environment with two identical data secure devices (actuators), for which I associated the same group object for input state object.

Now, I'm able to send the read request (and receive the answer) using the Group Monitoring on the ETS but, if I try the same functionality on my application, it does not work.

Here is the Calimero trace:

2023-07-21 08:59:22,951 TRACE c.?.1.1.136:3671 use tool access sequence 1
2023-07-21 08:59:22,957 DEBUG c.?.1.1.136:3671 sync tool access seq with 1.1.9
2023-07-21 08:59:22,958 DEBUG c.l.1.1.1.136:3671 send (wait for confirmation) 1.1.1->1.1.9 L_Data.req, system priority hop count 6 repeat, tpdu 03 f1 92 00 00 00 00 00 01 00 00 00 00 00 00 7b 62 d3 21 42 9f 91 b0 17 f6
2023-07-21 08:59:22,960 TRACE c.k.K.1.1.136:3671 sending cEMI frame seq 0, WaitForCon, attempt 1 (channel 2) 06 10 04 20 00 2c 04 02 00 00 11 00 30 60 11 01 11 09 18 03 f1 92 00 00 00 00 00 01 00 00 00 00 00 00 7b 62 d3 21 42 9f 91 b0 17 f6
2023-07-21 08:59:23,056 TRACE c.k.K.1.1.136:3671 from 192.168.1.136:3671: KNXnet/IP tunneling.req (0x420 v1.0) length 44: 06 10 04 20 00 2c 04 02 14 00 2e 00 30 60 11 01 11 09 18 03 f1 92 00 00 00 00 00 01 00 00 00 00 00 00 7b 62 d3 21 42 9f 91 b0 17 f6
2023-07-21 08:59:23,058 DEBUG c.k.K.1.1.136:3671 received request seq 20 (channel 2) cEMI L-Data.con 1.1.1->1.1.9
2023-07-21 08:59:23,060 DEBUG c.l.1.1.1.136:3671 confirmation of 1.1.9
2023-07-21 08:59:23,061 TRACE c.l.1.1.1.136:3671 send 1.1.1->1.1.9 succeeded
2023-07-21 08:59:23,157 TRACE c.k.K.1.1.136:3671 from 192.168.1.136:3671: KNXnet/IP tunneling.req (0x420 v1.0) length 44: 06 10 04 20 00 2c 04 02 15 00 29 00 30 60 11 09 11 01 18 03 f1 93 54 a4 64 1b d0 03 4e 10 8f ed 9f 55 b8 8f 68 30 89 64 22 27 54 db
2023-07-21 08:59:23,163 TRACE c.k.K.1.1.136:3671 received request seq 21 (channel 2) cEMI 29 00 30 60 11 09 11 01 18 03 f1 93 54 a4 64 1b d0 03 4e 10 8f ed 9f 55 b8 8f 68 30 89 64 22 27 54 db
2023-07-21 08:59:23,166 DEBUG c.?.1.1.136:3671 1.1.9->1.1.1 decrypt sync.res (auth+conf, tool access)
2023-07-21 08:59:23,171 INFO c.?.1.1.136:3671 tuwien.auto.calimero.secure.KnxSecureException: MAC mismatch 1.1.9->1.1.1

As you can see, the messages exchange is interrupted by that "MAC mismatch" error.
Please consider that I successfully import the keyring file (or at least it seems so) and the communication is successfully established with the tunneler.
Honestly, I don't know if the issue I'm facing is related to the library or it's something related to the configuration itself: if you can give me some advice it would be really appreciated!

If required, I can give you the ETS test project and some other detail related to the Calimero implementation on my application.

Thanks in advance and best regards!
Giorgio

bmalinowsky, 2023-08-04:
Hello,
you do receive a sync.res, so the key should be correct.
The sequence you see is actually synchronizing the tool seq number. Is the sync.res successfully shown in the ETS monitor?

Otherwise, you can either comment out the MAC computation and see if you get further, or check which tool seq the ETS uses, and init that value + 1 [here](https://github.com/calimero-project/calimero-core/blob/104731d015574d9aeb1089e3d6cbcf70c62d2b00/src/io/calimero/secure/SecureApplicationLayer.java#L380). This will then skip the sync step and directly do the read. Then you see if the read response is correctly handled or not.

calimero-project, 2023-09-19:
Any news on this?

closed 2024-02-06 by calimero-project