calimero-project/calimero-core #126: Can not connect to serial port in tpuart monitor on Android

aliglzr, 2024-02-07:
Hello, 
I want to monitor tpuart network but it can not open serial port.
I use below code to create tpuart network monitor:
```
    try (KNXNetworkMonitor knxMonitor = new KNXNetworkMonitorTpuart(portId, true)) {
            System.out.println("Connection established");

            knxMonitor.addMonitorListener(new LinkListener() {
                @Override
                public void indication(final FrameEvent e)
                {
                    System.out.println(e.getFrame() + ": " + ((MonitorFrameEvent) e).getRawFrame());
                }

                @Override
                public void linkClosed(CloseEvent e) {
                    System.out.println("Reason" + ":" + e.getReason());
                }
            });

            // let's wait some seconds to monitor KNX frames
            Thread.sleep(10000);
        } catch (InterruptedException e) {
            Log.d("Interrupted", e.getMessage());
        } catch (KNXException e) {
            e.printStackTrace();
        }
```
but when I run the application it gives me this error:
```
tuwien.auto.calimero.KNXException: no serial adapter available to open /dev/ttyS4
at tuwien.auto.calimero.serial.LibraryAdapter.open(LibraryAdapter.java:169)
at tuwien.auto.calimero.serial.TpuartConnection.<init>(TpuartConnection.java:183)
at tuwien.auto.calimero.link.KNXNetworkMonitorTpuart.<init>(KNXNetworkMonitorTpuart.java:71)
at com.example.myapplication.MainActivity2.onCreate(MainActivity2.java:53)
at android.app.Activity.performCreate(Activity.java:8072)
at android.app.Activity.performCreate(Activity.java:8052)
at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1341)
at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3688)
at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3864)
at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:103)
at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2253)
at android.os.Handler.dispatchMessage(Handler.java:106)
at android.os.Looper.loopOnce(Looper.java:201)
at android.os.Looper.loop(Looper.java:288)
at android.app.ActivityThread.main(ActivityThread.java:7870)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
```

I used another serial port library to connect to serial port and it works so I think the error is for calimero-core library
Can you help me to fix it?
Thank you in advance for your answers!

bmalinowsky, 2024-02-07:
You are missing an adapter for serial I/O, e.g., [calimero-rxtx](https://github.com/calimero-project/calimero-rxtx) for RXTX as well as the RXTX library. Put those two libraries on the classpath, then calimero should be able to open the port.

aliglzr, 2024-02-08:
I appreciate your reply!
I did what you said, and implemented this adapter:
`    implementation("com.github.calimero:calimero-rxtx:2.5")`
but now it gives me another error:
java.lang.NoClassDefFoundError: gnu.io.RXTXCommDriver thrown while loading gnu.io.RXTXCommDriver
```
I  Rejecting re-init on previously-failed class java.lang.Class<gnu.io.RXTXCommDriver>: java.lang.ExceptionInInitializerError: 
at java.lang.Class java.lang.Class.classForName(java.lang.String, boolean, java.lang.ClassLoader) (Class.java:-2)
at java.lang.Class java.lang.Class.forName(java.lang.String, boolean, java.lang.ClassLoader) (Class.java:454)
at java.lang.Class java.lang.Class.forName(java.lang.String) (Class.java:379)
at void gnu.io.CommPortIdentifier.<clinit>() (CommPortIdentifier.java:109)
at java.util.Enumeration gnu.io.CommPortIdentifier.getPortIdentifiers() (CommPortIdentifier.java:318)
at void tuwien.auto.calimero.serial.RxtxAdapter.open(java.lang.String, int) (RxtxAdapter.java:156)
at void tuwien.auto.calimero.serial.RxtxAdapter.<init>(org.slf4j.Logger, java.lang.String, int) (RxtxAdapter.java:100)
at java.lang.Object java.lang.reflect.Constructor.newInstance0(java.lang.Object[]) (Constructor.java:-2)
at java.lang.Object java.lang.reflect.Constructor.newInstance(java.lang.Object[]) (Constructor.java:343)
at tuwien.auto.calimero.serial.LibraryAdapter tuwien.auto.calimero.serial.LibraryAdapter.open(org.slf4j.Logger, java.lang.String, int, int) (LibraryAdapter.java:151)
at void tuwien.auto.calimero.serial.TpuartConnection.<init>(java.lang.String, java.util.Collection) (TpuartConnection.java:183)
at void tuwien.auto.calimero.link.KNXNetworkMonitorTpuart.<init>(java.lang.String, boolean) (KNXNetworkMonitorTpuart.java:71)
at void com.example.myapplication.MainActivity2.onCreate(android.os.Bundle) (MainActivity2.java:40)
at void android.app.Activity.performCreate(android.os.Bundle, android.os.PersistableBundle) (Activity.java:8072)
at void android.app.Activity.performCreate(android.os.Bundle) (Activity.java:8052)
at void android.app.Instrumentation.callActivityOnCreate(android.app.Activity, android.os.Bundle) (Instrumentation.java:1341)
at android.app.Activity android.app.ActivityThread.performLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent) (ActivityThread.java:3688)
at android.app.Activity android.app.ActivityThread.handleLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.app.servertransaction.PendingTransactionActions, android.content.Intent) (ActivityThread.java:3864)
at void android.app.servertransaction.LaunchActivityItem.execute(android.app.ClientTransactionHandler, android.os.IBinder, android.app.servertransaction.PendingTransactionActions) (LaunchActivityItem.java:103)
at void android.app.servertransaction.TransactionExecutor.executeCallbacks(android.app.servertransaction.ClientTransaction) (TransactionExecutor.java:135)
at void android.app.servertransaction.TransactionExecutor.execute(android.app.servertransaction.ClientTransaction) (TransactionExecutor.java:95)
at void android.app.ActivityThread$H.handleMessage(android.os.Message) (ActivityThread.java:2253)
at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:106)
at boolean android.os.Looper.loopOnce(android.os.Looper, long, int) (Looper.java:201)
at void android.os.Looper.loop() (Looper.java:288)
at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:7870)
at java.lang.Object java.lang.reflect.Method.invoke(java.lang.Object, java.lang.Object[]) (Method.java:-2)
at void com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run() (RuntimeInit.java:548)
at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:1003)
2024-02-08 17:48:40.508  3893-3893  System.err              com.example.myapplication            W  java.lang.NoClassDefFoundError: gnu.io.RXTXCommDriver thrown while loading gnu.io.RXTXCommDriver
2024-02-08 17:48:40.509  3893-3893  e.myapplicatio          com.example.myapplication            I  Rejecting re-init on previously-failed class java.lang.Class<gnu.io.RXTXCommDriver>: java.lang.ExceptionInInitializerError: 
at java.lang.Class java.lang.Class.classForName(java.lang.String, boolean, java.lang.ClassLoader) (Class.java:-2)
at java.lang.Class java.lang.Class.forName(java.lang.String, boolean, java.lang.ClassLoader) (Class.java:454)
at java.lang.Class java.lang.Class.forName(java.lang.String) (Class.java:379)
at void gnu.io.CommPortIdentifier.<clinit>() (CommPortIdentifier.java:109)
at java.util.Enumeration gnu.io.CommPortIdentifier.getPortIdentifiers() (CommPortIdentifier.java:318)
at void tuwien.auto.calimero.serial.RxtxAdapter.open(java.lang.String, int) (RxtxAdapter.java:156)
at void tuwien.auto.calimero.serial.RxtxAdapter.<init>(org.slf4j.Logger, java.lang.String, int) (RxtxAdapter.java:100)
at java.lang.Object java.lang.reflect.Constructor.newInstance0(java.lang.Object[]) (Constructor.java:-2)
at java.lang.Object java.lang.reflect.Constructor.newInstance(java.lang.Object[]) (Constructor.java:343)
at tuwien.auto.calimero.serial.LibraryAdapter tuwien.auto.calimero.serial.LibraryAdapter.open(org.slf4j.Logger, java.lang.String, int, int) (LibraryAdapter.java:151)
at void tuwien.auto.calimero.serial.TpuartConnection.<init>(java.lang.String, java.util.Collection) (TpuartConnection.java:183)
[... cut ...]
at android.app.ActivityThread.main(ActivityThread.java:7870)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
Caused by: gnu.io.NoSuchPortException
at gnu.io.CommPortIdentifier.getPortIdentifier(CommPortIdentifier.java:274)
at tuwien.auto.calimero.serial.RxtxAdapter.open(RxtxAdapter.java:162)
... 23 more
```
I think it does not find the appropriate library.
Thanks for your support! :)

bmalinowsky, 2024-02-09:
calimero-rxtx configures as dependency the RXTX library `com.neuronrobotics:nrjavaserial:5.2.1`. That library also contains `gnu.io.RXTXCommDriver` which is said to be missing in your stacktrace. Do you have `nrjavaserial` on your classpath?

aliglzr, 2024-02-10:
Hello dear bmalinowsky,
Yes, I have `nrjavaserial` in my project classpath.
The `calimero-rxtx` have imported this library itself.
But I imported this library manually and the error still persists.
Thanks for your support.

aliglzr, 2024-02-10:
I think `nrjavaserial` does not support android because in the apk file, native libraries for android does not exist.
Is there any rxtx adapter with another library that supports android for `calimero-core`?

bmalinowsky, 2024-02-18:
Yes, from what I see `nrjavaserial` dropped support for android due to permission changes in android 4.x.

I have another provider [serial-native](https://github.com/calimero-project/serial-native), which has a native implementation in C for serial communication. I tried the 3.0-SNAPSHOT on branch _android_ and it seem to work with minor modifications in the emulator. I cannot test actual communication though, because I don't have an android device with serial port.

You would have to adjust the setup for serial-native v2.5.

aliglzr, 2024-02-20:
Thank you very much!
You solved my problem and now it works using `serial-native` library.
Is it possible to put your crypto currency wallet here so that I can donate an amount to you?
I really appreciate your help.
Thank you for this very complete and practical library that you developed.

closed 2024-02-26 by aliglzr