calimero-project/calimero-core #127: DPT 9.x lower boundary

holgerfriedrich, 2024-02-24:
During implementation of integration tests, I experienced a problem with the lower boundary:

Frame raw data 0xf800 should correspond to the lower boundary (at least for those subtypes not defined from 0 or -273°C).
Range according to spec is -671088,64 … 670433,28.

Decoding is fine. 0xf800 results in -671088,64.

However, encoding the same value does not give 0xf800, but results in exception:
tuwien.auto.calimero.KNXFormatException: 9.011 Time difference 2: translation error, value out of range [-670760..+670760]: -671088.64

I know this should not have real practical relevance, so please feel free to close my issue as "won't fix".

bmalinowsky, 2024-03-03:
Where exactly did you look this up? In v1.8.3, DPT 9.011 has specified a neg. limit of -670 760 ms. So I think it should be the opposite, that decoding should also enforce that check and complain.

holgerfriedrich, 2024-03-03:
KNX support posted a newer version of the spec...
https://support.knx.org/hc/en-us/article_attachments/15392631105682

The overall range for seem to have changed, though I never noticed in the changelog of the document :-(

bmalinowsky, 2024-03-03:
Right, even in the 2019 release (it has the same version number, though) it was already changed. I never noticed that. Thanks for pointing it out! I will update the limits.

commit 844be8a4 (2024-03-03): Update dpt 9.x range limits to dpt spec v2.2.1 (closes #127)
closed 2024-03-03 by bmalinowsky