calimero-project/calimero-core #128: DPT 7.003 and 7.004 unit

holgerfriedrich, 2024-02-24:
It seems that DPT 7.003 and 7.004 have unit "ms" in the spec, although the example range is given in seconds.

Calimero uses "ms" for DPT 7.002, and "s" for 7.003 and 7.004.

Proposal to change to "ms" and adapt the boundary as well.

Please feel free to close this issue if you think this should not be changed.

bmalinowsky, 2024-03-03:
Yes, it makes sense to go with the unit as specified in the dedicated column, and not the unit used for specifying the range.

bmalinowsky, 2024-03-03:
@holgerfriedrich FYI, this will also likely affect DPT 243.600, which uses 7.004, and is currently in [s].

holgerfriedrich, 2024-03-03:
@bmalinowsky just checkd my unit tests, time component of 243.600 seems to be fine.

bmalinowsky, 2024-03-04:
@holgerfriedrich thanks for for reports!

commit 3bcd36af (2024-03-04): Change unit of dpt 7.003 and 7.004 to ms (fixes #128)
closed 2024-03-04 by bmalinowsky