calimero-project/calimero-core #129: Some missing DPT subtypes 20.x and 21.x

holgerfriedrich, 2024-03-12:
DPTs 20 and 21 are currently not fully covered, as new subtypes have been added to the specification.

The following subtypes seem to be missing:
DPT 20: 611, 612, 613, 1203, 1204, 1205, 1206, 1207, 1208, 1209
DPT 21.1200 and 21.1201.

This is not urgent, just to let you know about the gaps identified.

bmalinowsky, 2024-03-17:
@holgerfriedrich Thanks for the update! I put them into branches 
[feat/new-dpts-20.x](https://github.com/calimero-project/calimero-core/tree/feat/new-dpts-20.x) and 
[feat/new-dpts-21.x](https://github.com/calimero-project/calimero-core/tree/feat/new-dpts-21.x). 
So these DPTs will already be in 2.6.

closed 2024-03-17 by bmalinowsky

holgerfriedrich, 2024-03-17:
Good news, thank you!
Any new release will be appreciated :+1: 

What are your plans regarding 2.6 and 3.0? Can we expect anything the next months?

The last days, I have submitted a patch to make openHAB make with 3.0-SNAPSHOT, openhab/openhab-addons#16506

bmalinowsky, 2024-03-27:
@holgerfriedrich 2.6-rc1 is out, I will let that version settle a little to see if something odd comes up. 
3.0 will get a milestone release, to see if the modularization (and along with that namespace changes, also logger changed) are OK, and if there are problems if someone uses a modularized setup. 

Version 2.6 should contain all commits from master, sans modularization/namespace/logger which goes into 3.0.