calimero-project/calimero-core #137: Can't compiled in Android Java17

swFan, 2024-10-16:
when i compile calimero-core using Java 17 in android, it does not works.  following compile info:

G:\TCSSoft\AndroidProjects\Calimero\calimero-core-v26rc1\src\main\java\tuwien\auto\calimero\serial\ConnectionFactory.java:43: 错误: 找不到符号
import java.util.ServiceLoader.Provider;
                              ^
  符号:   类 Provider
  位置: 类 ServiceLoader
G:\TCSSoft\AndroidProjects\Calimero\calimero-core-v26rc1\src\main\java\tuwien\auto\calimero\serial\ConnectionFactory.java:113: 错误: 找不到符号
	private Stream<Provider<P>> providers(final boolean refresh) {
	               ^
  符号:   类 Provider
  位置: 类 ConnectionFactory<P,C>
  其中, P,C是类型变量:
    P扩展已在类 ConnectionFactory中声明的Object
    C扩展已在类 ConnectionFactory中声明的Object
G:\TCSSoft\AndroidProjects\Calimero\calimero-core-v26rc1\src\main\java\tuwien\auto\calimero\serial\ConnectionFactory.java:110: 错误: 不兼容的类型: Stream<Object>无法转换为Stream<P>
		}).filter(Objects::nonNull);
		         ^
  其中, P是类型变量:
    P扩展已在类 ConnectionFactory中声明的Object

swFan, 2024-10-16:
[image]

bmalinowsky, 2024-10-16:
You need Android API level 35.

closed 2024-11-16 by calimero-project