calimero-project/calimero-core #138: Support for knx secure over RF

bhanujarya, 2024-11-25:
Hi Channel i am looking to use knx secure over RF can you suggest if support for the same is available along with any reference implementation example for the same.
Link for the product: https://weinzierl.de/en/products/knx-rf-usb-interface-stick-340/
Link for the SDK: https://weinzierl.de/en/software/knx/sdk-for-baos-binary-services/

I have successfully loaded the required keywring file and password .

When i am writing the command to relay device 
```
	public static void relayOn(ProcessCommunicatorImpl pc, GroupAddress relayAddress) {
		try {
			System.out.println("Inside relay on method ");
			DPTXlatorBoolean onValue = new DPTXlatorBoolean(DPTXlatorBoolean.DPT_BOOL);
			onValue.setValue(true);
			System.out.println("RElay address is " + relayAddress + " on value " + onValue);
			pc.responseTimeout(Duration.ofSeconds(15));

			pc.write(relayAddress, onValue);
		} catch (KNXException e) {
			System.err.println("Exception : on value " + e);
			e.printStackTrace();
		}
	}
```

I am getting 
****WARNING: waiting for GO diagnostics****
```
security done 
Nov 25, 2024 7:12:17 PM io.calimero.usb.provider.javax.UsbConnection <clinit>
INFO: Found 1 KNX USB devices:
 |--Bus 1 Device 9: ID 0e77:2001 Weinzierl Engineering GmbH KNX-USB Interface S/N 00C50105B3AA
 |   Attached at port 1 (bus:1/1)
 |   Full Speed USB
KNX Secure is ready
Inside relay on method 
RElay address is 1/1/1 on value DPT 1.002 [true]
Nov 25, 2024 7:12:21 PM io.calimero.process.ProcessCommunicatorImpl send
WARNING: waiting for GO diagnostics
java.util.concurrent.TimeoutException
	at java.base/java.util.concurrent.CompletableFuture$Timeout.run(CompletableFuture.java:2874)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:833)
```
I am able to write to knx secure rf device using ETS Utiltiy. Can you please help in resolution steps for the above issue and any code correction if required

bmalinowsky, 2024-12-01:
Maybe you need to supply the RF domain of your RF network to `new RFSetttings(...)`.

bhanujarya, 2024-12-04:
Hi @bmalinowsky , i have actually tried both the ways by supplying the domain and without domain , but i am getting
RElay address is 1/1/1 on value DPT 1.002 [true]
Nov 25, 2024 7:12:21 PM io.calimero.process.ProcessCommunicatorImpl send
```
WARNING: waiting for GO diagnostics
java.util.concurrent.TimeoutException
	at java.base/java.util.concurrent.CompletableFuture$Timeout.run(CompletableFuture.java:2874)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:833)
```
I checked this was being used in security layer library of knx , can you please help in how can we pass through security layer and actually communicated with my secure light .

bhanujarya, 2024-12-06:
Hi @bmalinowsky , i have got the rf address as **00FA:F4E3725F** , i have passed it code 
       String rfDomain = "00FA:F4E3725F";
```

        byte[] rfDomainBytes = parseRFDomain(rfDomain);

		IndividualAddress dev = new IndividualAddress(rfDomainBytes);

		try (var knxLink = new KNXNetworkLinkUsb(device, new RFSettings(dev)); // This process communicator constructor
				
				var pc = new ProcessCommunicatorImpl(knxLink)) {

			addProccessListners(pc);
			

			System.out.println("KNX Secure is ready");
```
But still i am getting timeout execption 
```

Exception in thread "main" io.calimero.secure.KnxSecureException: sync.req with 1.1.1
	at io.calimero.secure.SecureApplicationLayer.syncWith(SecureApplicationLayer.java:943)
	at io.calimero.secure.SecureApplicationLayer.secureData(SecureApplicationLayer.java:383)
	at io.calimero.secure.SecureApplicationLayer.writeGroupObjectDiagnostics(SecureApplicationLayer.java:305)
	at io.calimero.process.ProcessCommunicatorImpl.send(ProcessCommunicatorImpl.java:481)
	at io.calimero.process.ProcessCommunicatorImpl.write(ProcessCommunicatorImpl.java:444)
	at io.calimero.process.ProcessCommunicatorImpl.write(ProcessCommunicatorImpl.java:375)
	at knxtest.KNXRFBackup.relayOn(KNXRFBackup.java:172)
	at knxtest.KNXRFBackup.main(KNXRFBackup.java:89)
Caused by: java.util.concurrent.TimeoutException
	at java.base/java.util.concurrent.CompletableFuture$Timeout.run(CompletableFuture.java:2874)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:833)
```

Please help me in solving this issue with corrected code

closed 2025-04-18 by calimero-project