calimero-project/calimero-core #139: Calimaro 3.0 Snapshot Dependency Issue

bhanujarya, 2024-11-25:
Hi Channel i am looking to use knx secure over RF can you suggest if support for the same is available along with any reference implementation example for the same.
Link for the product: https://weinzierl.de/en/products/knx-rf-usb-interface-stick-340/
Link for the SDK: https://weinzierl.de/en/software/knx/sdk-for-baos-binary-services/
To process the code i needed to use dependency
`import io.calimero.*;`
This dependency is only available in Calimaro 3.0 Snapshot . But in the global maven repo latest maven dependency is
```
<dependency>
    <groupId>com.github.calimero</groupId>
    <artifactId>calimero-core</artifactId>
    <version>2.6-rc1</version>
</dependency>
```
Can you help in correct maven dependency .Calimaro 3.0 snapshot dependency available through maven but not gradle..

bmalinowsky, 2024-12-01:
The snapshot repository for Maven artifacts is _https://s01.oss.sonatype.org/content/repositories/snapshots_.

holgerfriedrich, 2024-12-04:
I think what you are missing is the definition of the repo. It can be done like this:

```
  <repositories>
    <repository>
      <id>sonatype.snapshots</id>
      <name>https://s01.oss.sonatype.org/content/repositories/snapshot</name>
      <url>https://s01.oss.sonatype.org/content/repositories/snapshots/</url>
      <releases>
        <enabled>false</enabled>
      </releases>
      <snapshots>
        <enabled>true</enabled>
      </snapshots>
    </repository>
  </repositories>

  <dependencies>
    <dependency>
      <groupId>io.calimero</groupId>
      <artifactId>calimero-core</artifactId>
      <version>3.0-SNAPSHOT</version>
    </dependency>
  </dependencies>
```

bhanujarya, 2024-12-05:
Thank you very much @holgerfriedrich  for the dependency , it is correct and helping me with imports
I have checked it with all the the imports and the dependency actually works with 
```
import io.calimero.DataUnitBuilder;
import io.calimero.FrameEvent;
import io.calimero.KNXException;
import io.calimero.knxnetip.Discoverer;
import io.calimero.knxnetip.KNXnetIPConnection;
import io.calimero.knxnetip.SecureConnection;
import io.calimero.knxnetip.TcpConnection;
import io.calimero.link.KNXLinkClosedException;
import io.calimero.link.KNXNetworkLink;
import io.calimero.link.KNXNetworkLinkIP;
import io.calimero.link.KNXNetworkLinkUsb;
import io.calimero.link.NetworkLinkListener;
import io.calimero.link.medium.KnxIPSettings;
import io.calimero.link.medium.RFSettings;
import io.calimero.link.medium.TPSettings;
import io.calimero.GroupAddress;
import io.calimero.IndividualAddress;
import io.calimero.dptxlator.DPTXlator2ByteFloat;
import io.calimero.dptxlator.DPTXlator4ByteFloat;
import io.calimero.dptxlator.DPTXlatorBoolean;
import io.calimero.process.ProcessCommunicator;
import io.calimero.process.ProcessCommunicatorImpl;
import io.calimero.process.ProcessEvent;
import io.calimero.process.ProcessListener;
import io.calimero.secure.Keyring;
import io.calimero.secure.Security;
import io.calimero.serial.ConnectionFactory.ThrowingFunction;
import io.calimero.serial.usb.UsbConnectionFactory;
import io.calimero.Priority;
import io.calimero.DetachEvent;
```

But just one import is not working with the above dependency 
`import io.calimero.device.ProcessCommunicationResponder;
`
Can you help in sharing the neceesary maven dependency for this .

holgerfriedrich, 2024-12-05:
You could try to add a second dependency, calimero-device

bhanujarya, 2024-12-07:
Thanks @holgerfriedrich for the support

closed 2024-12-23 by bhanujarya