calimero-project/calimero-core #145: `ConnectionBase.send` does not return on `KNXAckTimeoutException`

Darthyson, 2025-01-11:
To begin with, many thanks for this great project and the effort that went into it.

One of our users has a Loxone Miniserver Gen1 as KNX IP interface.
For whatever reason, it returns the tunnel ack with status 0x04 for every tunnel request with sequence number 255.
[image]
[wireshark_loxone_error_0x04_on_seqnmb255.zip](https://github.com/user-attachments/files/18387032/wireshark_loxone_error_0x04_on_seqnmb255.zip)

To reproduce this behavior on my KNX test bus, I modified the current knxd/knxd@477ab2be192a12b816b2bb9d0a6025ccb6c32c24 debian branch to set the tunnel ack status to 0x04 for the sequence number I specify.
This is my simple test program which connects to the device `15.15.192` and tries to writes 2 bytes into its memory:
```java
public class Main {
    public static void main(String[] args) {
        Logger logger = (Logger) LoggerFactory.getLogger(Main.class);
        InetSocketAddress local = new InetSocketAddress("0.0.0.0", 0);
        InetSocketAddress remote = new InetSocketAddress("192.168.0.222", 3671);
        IndividualAddress device = new IndividualAddress(15,15,192);
        try {
            KNXNetworkLink link = KNXNetworkLinkIP.newTunnelingLink(local, remote, false, new TPSettings());
            ManagementClientImpl mc = new ManagementClientImpl(link);
            Destination d = mc.createDestination(device, true, false, false);
            byte[] testBytes = new byte[]{(byte) 0xbe, (byte) 0xef};
            for (int i = 0; i < 300; i++) {
                logger.warn("send memWrite #{}", i);
                mc.writeMemory(d, 0x3010, testBytes);
                logger.warn("done memWrite #{}", i);
            }
            logger.warn("cleaning up");
            mc.close();
            link.close();
            logger.warn("cleanup done");
        }
        catch (KNXTimeoutException e) {
            logger.error("got it {} -> {}", e.getClass(), e.getMessage());
        }
        catch (KNXException | InterruptedException e) {
            logger.error("{} -> {}", e.getClass(), e.getMessage());
        }
    }
}
```
Running this with [calimero-core/release/2.6](https://github.com/calimero-project/calimero-core/commit/c3a02eaaa3a8db160561e9bd20e3e23810d2d130) and knxd set to send status `0x04` on `sequence number 2`, I get following trace logs:
[wireshark_knxd_error_0x04_on_seqnmb2.zip](https://github.com/user-attachments/files/18387017/wireshark_knxd_error_0x04_on_seqnmb2.zip)
[testacktimeout-2025-01-11_20-13-44.317.log](https://github.com/user-attachments/files/18387027/testacktimeout-2025-01-11_20-13-44.317.log)
[testacktimeout-2025-01-11_20-13-44.317_html.zip](https://github.com/user-attachments/files/18387039/testacktimeout-2025-01-11_20-13-44.317_html.zip)

this callstack:
```
wait:-1, Object (java.lang)
join:1313, Thread (java.lang)
join:1381, Thread (java.lang)
quit:158, EventNotifier (tuwien.auto.calimero.link)
connectionClosed:115, EventNotifier (tuwien.auto.calimero.link)
connectionClosed:232, AbstractLink$LinkNotifier (tuwien.auto.calimero.link)
lambda$fireConnectionClosed$1:600, ConnectionBase (tuwien.auto.calimero.knxnetip)
accept:-1, ConnectionBase$$Lambda$220/0x000001765711f268 (tuwien.auto.calimero.knxnetip)
fire:143, EventListeners (tuwien.auto.calimero.internal)
fireConnectionClosed:600, ConnectionBase (tuwien.auto.calimero.knxnetip)
cleanup:504, ConnectionBase (tuwien.auto.calimero.knxnetip)
cleanup:261, ClientConnection (tuwien.auto.calimero.knxnetip)
close:492, ConnectionBase (tuwien.auto.calimero.knxnetip)
send:262, ConnectionBase (tuwien.auto.calimero.knxnetip)
send:212, KNXnetIPTunnel (tuwien.auto.calimero.knxnetip)
doSend:514, KNXNetworkLinkIP (tuwien.auto.calimero.link)
onSend:502, KNXNetworkLinkIP (tuwien.auto.calimero.link)
send:406, AbstractLink (tuwien.auto.calimero.link)
sendRequestWait:479, KNXNetworkLinkIP (tuwien.auto.calimero.link)
sendData:370, TransportLayerImpl (tuwien.auto.calimero.mgmt)
send:1421, ManagementClientImpl (tuwien.auto.calimero.mgmt)
send:1251, ManagementClientImpl (tuwien.auto.calimero.mgmt)
writeMemory:1159, ManagementClientImpl (tuwien.auto.calimero.mgmt)
main:29, Main (org.selfbus.testacktimeout)
```
and this [threads_report.txt](https://github.com/user-attachments/files/18387148/threads_report.txt)

Since I'm quite unfamiliar with threading in Java, I don't know where to look for the cause that `ConnectionBae.send` never returns on an `ACK_ERROR`.
I did also a short test with the [current master](https://github.com/calimero-project/calimero-core/commit/dea3e89e733ed3c4232ed280548ce482495dd58c) with same result.

bmalinowsky, 2025-01-25:
Thank you for your report! 
This is an issue with the layer 4 `ackLock` waiting to acquire the lock for cleanup (which is still held by the sending TL): https://github.com/calimero-project/calimero-core/blob/dea3e89e733ed3c4232ed280548ce482495dd58c/src/io/calimero/mgmt/TransportLayerImpl.java#L289

I will look into how to fix this bug.

Darthyson, 2025-02-05:
Commit c1e6658 does not completely solve the problem.
For some sequence numbers (0, 1, 2, 4, ..., 254) it seems to work fine :thumbsup:.
For others (mostly odd) sequence numbers (3, 5, ..., 255) `ConnectionBase.send` still does not return.
Maybe this is `T_ACK` related?
Attached are some logs (incl. wireshark, stacktrace, threads_report) for sequence numbers 0-5 and 254, 255:
[issue_145_does_not_return_on_some_sequence_numbers.zip](https://github.com/user-attachments/files/18671552/issue_145_does_not_return_on_some_sequence_numbers.zip)

This is my updated test program with minor logging changes and usage of current master c1e6658:
[Main.java.zip](https://github.com/user-attachments/files/18671573/Main.java.zip)

bmalinowsky, 2025-02-05:
Thank you for testing! I did a quick look at your traces; I think it's still the same root cause, but different location. (In the _stacktrace_ .txt files the lock is in `waitForAck` now). In wireshark I see that calimero tries to do an orderly shutdown, because it now received twice the `TunnelAck ... Error 0x04`. So the notification thread again tries to wake anybody waiting for the `ackLock` during cleanup.

Btw, I updated the 2.6 release branch so it's up to par with the latest fixes from master (I see you have your code using the 2.6 version).

closed 2025-02-09 by bmalinowsky

Darthyson, 2025-02-10:
Thanks a lot. My quick tests with sequence numbers 0 to 11 and 244 to 255 (which return status 0x04) with b4114ab (release/2.6) were all successful 👍 .

bmalinowsky, 2025-02-10:
@Darthyson  Good to hear :)

commit c1e66589 (2025-02-01): Reduce scope of ackLock in sendData to eliminate deadlock possibility (fixes #145).

Notification thread calling TL::destroyDestination and waiting to acquire ackLock while TL::sendData waits in sendRequestWait would prevent cleanup to finish.
commit 8be28b37 (2025-02-09): Reduce scope of ackLock in waitForAck to eliminate deadlock possibility (fixes #145).
commit b4114abe (2025-02-09): Reduce scope of ackLock in waitForAck to eliminate deadlock possibility (fixes #145).