calimero-project/calimero-core #148: Differences between calimero-core 2.5.1 and V3

Alecocluc, 2025-04-02:
Hello! We were trying to implement calimero and noticed there is a V3 version, are there any more changes apart from changing the imports to io.calimero and new libraries like calimero-devices, calimero-usb, calimero-rxtx? Is it safe to use this snapshot at the moment? Because we would like to use some new features from this version. Thanks in advance!

bmalinowsky, 2025-04-03:
calimero-device and calimero-rxtx also exist for 2.x. If you look at the current RC, which is v2.6-rc2, there are almost no differences in the commits compared to v3. Deprecated stuff got removed etc. The main difference is that v3 is modularized. I don't think there are currently any new features in v3?

Alecocluc, 2025-04-03:
Oh, ok. Thanks for letting me know!!!

closed 2025-04-18 by calimero-project