calimero-project/calimero-core #149: Error in java.net.NetworkInterface when using calimero Discover in Kotlin for an Android APP

Arky222, 2025-04-04:
I'm currently using Calimero-Core 2.5.1 trying to implement some of its functions in an Android app written in Kotlin.

To implement the Discoverer, I simply copied and pasted the Introduction Discoverer class code:

```
        var foundNetworks: String = ""
        val useNAT = false
        Discoverer.udp(useNAT).timeout(Duration.ofSeconds(3)).search().get()
            .forEach {
                val res = it.response.toString().replace(", ", "\n\t")
                foundNetworks += ("${it.networkInterface.name} ${it.localEndpoint()} <=> $res")
            }
        return foundNetworks
```

However, when I compile and run the app on an Android device, I get the following error in logcat:

```
java.lang.NoSuchMethodError: No static method networkInterfaces()Ljava/util/stream/Stream; in class Ljava/net/NetworkInterface; or its super classes (declaration of 'java.net.NetworkInterface' appears in /apex/com.android.art/javalib/core-oj.jar)                                                                             
    at tuwien.auto.calimero.knxnetip.Discoverer.searchAsync(Discoverer.java:519)
    at tuwien.auto.calimero.knxnetip.Discoverer.search(Discoverer.java:355)
```

As far as I know, the problem here is that Android is based on an older version of Java and "NetworkInterface.networkInterfaces()" can't be called, so this function doesn't work properly:

```
private CompletableFuture<List<Result<SearchResponse>>> searchAsync(final Duration timeout,
			final Srp... searchParameters) {
		if (timeout.isNegative())
			throw new KNXIllegalArgumentException("timeout has to be >= 0");
		final NetworkInterface[] nifs;
		try {
			nifs = NetworkInterface.networkInterfaces().toArray(NetworkInterface[]::new);
		}
		catch (final SocketException e) {
			return CompletableFuture.failedFuture(e);
		}
```

I also faced this very same problem when trying to use `KNXNetworkLinkIP` since it also calls the same NetworkInterface function:

```
val knxNetwork = KNXNetworkLinkIP.newTunnelingLink(
       localSocketAddress, ipSocketAddress, false, config
)
```

Leaving localSocketAddress `null` launches the same error since Calimero looks for the local IP address on its own, but I managed a way to use `NetworkInterface.getNetworkInterfaces()` instead of `NetworkInterface.networkInterfaces` to find the localAddress myself.

I went through the Issues regarding this problem and found out an old answer saying that we should be using calimero-core 2.4 since its based on an older Java version (more suitable for Android indeed). So I have done a lot of testing and I managed to make it work but this Exception is thrown:

```
tuwien.auto.calimero.KNXException: search could not be started on any network interface
```

I guess this might be related to Java not being able to list Android networks somehow, despite `Discover.startSearch()` function being able to call `NetworkInterface.getNetworkInterfaces()` properly.

I also tried calling `discoverer.search()` to look for the network manually, but since it is a private function I can't do it.

Is there any way I can sort this problem out without modifying calimero-core myself? I know this library is meant for Java, but if there's any workaround I can do to get it working in Android, I'm open to try it.

bmalinowsky, 2025-04-04:
It is true that the Android project is rather slacking/erratic in efforts making recent Java source code work without modification. 

Please print a trace log. If you look at the source code (which is readily available to you), you will find that the exception might not mean what you've guessed.

Arky222, 2025-04-07:
This is what I get when I print Android Studio native stack trace. However, implementing SLF4J logger simply doesn't show anything.

```
interfacetuwien.auto.calimero.KNXException: search could not be started on any network interface
 at tuwien.auto.calimero.knxnetip.Discoverer.startSearch(Discoverer.java:454)
 at es.iknx.iknxschool.CalimeroScannerActivity.initScanButton$lambda$1(CalimeroScannerActivity.kt:43)
 at es.iknx.iknxschool.CalimeroScannerActivity.$r8$lambda$4PXlhcLQN8JAn5-WjYNZx2ZOdY4(Unknown Source:0)
 at es.iknx.iknxschool.CalimeroScannerActivity$$ExternalSyntheticLambda0.onClick(D8$$SyntheticClass:0)
 at android.view.View.performClick(View.java:8028)
 at com.google.android.material.button.MaterialButton.performClick(MaterialButton.java:1218)
 at android.view.View.performClickInternal(View.java:8005)
 at android.view.View.-$$Nest$mperformClickInternal(Unknown Source:0)
 at android.view.View$PerformClick.run(View.java:31229)
 at android.os.Handler.handleCallback(Handler.java:959)
 at android.os.Handler.dispatchMessage(Handler.java:100)
 at android.os.Looper.loopOnce(Looper.java:232)
 at android.os.Looper.loop(Looper.java:317)
 at android.app.ActivityThread.main(ActivityThread.java:8705)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:580)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:886)
```

This means the problem is `final List<ReceiverLoop> rcv = new ArrayList<>();` size equals 0... but I don't get what that means. Does it mean there are no KNX networks available?

Arky222, 2025-04-07:
This is what I get when I print Android Studio native stack trace. However, implementing SLF4J logger simply doesn't show anything.

```
interfacetuwien.auto.calimero.KNXException: search could not be started on any network interface
 at tuwien.auto.calimero.knxnetip.Discoverer.startSearch(Discoverer.java:454)
 at es.CalimeroScannerActivity.initScanButton$lambda$1(CalimeroScannerActivity.kt:43)
 at es.CalimeroScannerActivity.$r8$lambda$4PXlhcLQN8JAn5-WjYNZx2ZOdY4(Unknown Source:0)
 at es.CalimeroScannerActivity$$ExternalSyntheticLambda0.onClick(D8$$SyntheticClass:0)
 at android.view.View.performClick(View.java:8028)
 at com.google.android.material.button.MaterialButton.performClick(MaterialButton.java:1218)
 at android.view.View.performClickInternal(View.java:8005)
 at android.view.View.-$$Nest$mperformClickInternal(Unknown Source:0)
 at android.view.View$PerformClick.run(View.java:31229)
 at android.os.Handler.handleCallback(Handler.java:959)
 at android.os.Handler.dispatchMessage(Handler.java:100)
 at android.os.Looper.loopOnce(Looper.java:232)
 at android.os.Looper.loop(Looper.java:317)
 at android.app.ActivityThread.main(ActivityThread.java:8705)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:580)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:886)
```

This means the problem is `final List<ReceiverLoop> rcv = new ArrayList<>();` size equals 0... but I don't get what that means. Does it mean there are no KNX networks available?

closed 2025-05-21 by Arky222