calimero-project/calimero-core #154: 我想用go调用咱们的库

bert1020, 2025-05-20:
计划使用GraalVM编译出来这个库,主要使用里面的USB接口,然后使用go的cgo调用,这个想法可行吗?因为我了解到GraalVM不支持反射和动态

bert1020, 2025-05-21:
我尝试了,发现IP路由设备是可以的,但是发现USB设备报错了,直接在java中运行是可以的,使用GraalVM编译出来之后,调用报以下错误,可以帮我解决一下吗?
```io.calimero.usb.provider.javax.UsbConnection <clinit>
严重: Enumerate USB devices, java.lang.UnsupportedOperationException: null not supported as key!
Unhandled exception: java.lang.UnsupportedOperationException: null not supported as key!
    at org.graalvm.collections.EconomicMapImpl.checkKeyNonNull(EconomicMapImpl.java:640)
    at org.graalvm.collections.EconomicMapImpl.get(EconomicMapImpl.java:242)
    at com.oracle.svm.core.jni.access.JNIReflectionDictionary.getDeclaredMethod(JNIReflectionDictionary.java:245)
    at com.oracle.svm.core.jni.access.JNIReflectionDictionary.findMethod(JNIReflectionDictionary.java:210)
    at com.oracle.svm.core.jni.access.JNIReflectionDictionary.getMethodID(JNIReflectionDictionary.java:255)
    at com.oracle.svm.core.jni.functions.JNIFunctions$Support.getMethodID(JNIFunctions.java:1842)
    at com.oracle.svm.core.jni.functions.JNIFunctions.ThrowNew(JNIFunctions.java:908)
    at org.usb4java.LibUsb.init(LibUsb.java:-2)
    at io.calimero.usb.provider.javax.UsbConnection.getDevicesLowLevel(UsbConnection.java:902)
    at io.calimero.usb.provider.javax.UsbConnection.attachedKnxDevices(UsbConnection.java:312)
    at io.calimero.usb.provider.javax.UsbConnectionProvider.attachedKnxUsbDevices(UsbConnectionProvider.java:61)
    at java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:215)
    at java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:197)
    at java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:215)
    at java.util.ServiceLoader$ProviderSpliterator.tryAdvance(ServiceLoader.java:1352)
    at java.util.Spliterator.forEachRemaining(Spliterator.java:332)
    at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:570)
    at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:560)
    at java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:921)
    at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:265)
    at java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:727)
    at io.calimero.serial.usb.UsbConnectionFactory.attachedKnxUsbDevices(UsbConnectionFactory.java:75)
    at org.example.NativeLibrary.discoverKnxUSBServers(NativeLibrary.java:48)

bmalinowsky, 2025-05-21:
You should be able for `DeviceInfo` to `--initialize-at-build-time`, as static init for that class does nothing.

bert1020, 2025-05-22:
> 我尝试了,发现IP路由设备是可以的,但是发现USB设备报错了,直接在java中运行是可以的,使用GraalVM编译出来之后,调用报以下错误,可以帮我解决一下吗?
> 
> ```
> 严重: Enumerate USB devices, java.lang.UnsupportedOperationException: null not supported as key!
> Unhandled exception: java.lang.UnsupportedOperationException: null not supported as key!
>     at org.graalvm.collections.EconomicMapImpl.checkKeyNonNull(EconomicMapImpl.java:640)
>     at org.graalvm.collections.EconomicMapImpl.get(EconomicMapImpl.java:242)
>     at com.oracle.svm.core.jni.access.JNIReflectionDictionary.getDeclaredMethod(JNIReflectionDictionary.java:245)
>     at com.oracle.svm.core.jni.access.JNIReflectionDictionary.findMethod(JNIReflectionDictionary.java:210)
>     at com.oracle.svm.core.jni.access.JNIReflectionDictionary.getMethodID(JNIReflectionDictionary.java:255)
>     at com.oracle.svm.core.jni.functions.JNIFunctions$Support.getMethodID(JNIFunctions.java:1842)
>     at com.oracle.svm.core.jni.functions.JNIFunctions.ThrowNew(JNIFunctions.java:908)
>     at org.usb4java.LibUsb.init(LibUsb.java:-2)
>     at io.calimero.usb.provider.javax.UsbConnection.getDevicesLowLevel(UsbConnection.java:902)
>     at io.calimero.usb.provider.javax.UsbConnection.attachedKnxDevices(UsbConnection.java:312)
>     at io.calimero.usb.provider.javax.UsbConnectionProvider.attachedKnxUsbDevices(UsbConnectionProvider.java:61)
>     at java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:215)
>     at java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:197)
>     at java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:215)
>     at java.util.ServiceLoader$ProviderSpliterator.tryAdvance(ServiceLoader.java:1352)
>     at java.util.Spliterator.forEachRemaining(Spliterator.java:332)
>     at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:570)
>     at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:560)
>     at java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:921)
>     at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:265)
>     at java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:727)
>     at io.calimero.serial.usb.UsbConnectionFactory.attachedKnxUsbDevices(UsbConnectionFactory.java:75)
>     at org.example.NativeLibrary.discoverKnxUSBServers(NativeLibrary.java:48)
> ```

应该解决这个错误,因为这个是编译成功的,go调用的时候报的错,我通过在编译的时候,加了某些参数,导致后面一直编译不成功.仔细看日志好像是LibUsb.init返回了-2 导致的 ,为啥会返回-2,我觉得解决这个应该就可以用了

closed 2025-05-22 by bert1020