calimero-project/calimero-core #155: unknown mask version 00FA

bert1020, 2025-06-06:
连接USB设备时报以下错误
```
io.calimero.KNXIllegalArgumentException: unknown mask version 00FA
	at io.calimero.DeviceDescriptor$DD0.from(DeviceDescriptor.java:199)
	at io.calimero.usb.provider.javax.UsbConnection.deviceDescriptor(UsbConnection.java:419)
	at io.calimero.link.KNXNetworkLinkUsb.<init>(KNXNetworkLinkUsb.java:142)
	at io.calimero.link.KNXNetworkLinkUsb.<init>(KNXNetworkLinkUsb.java:119)
```

这是搜索出来的设备信息
```
信息: Found 1 KNX USB devices:
 |--Bus 1 Device 3: ID 0403:6898 TSY Technology TSY KNX-USB Data Interface S/N 6587EEA40336
 |   Attached at port 4 (bus:1/1/4)
 |   Full Speed USB
```

bmalinowsky, 2025-06-06:
This is the same as your [other issue](https://github.com/calimero-project/calimero-server/issues/26).

bert1020, 2025-06-09:
不好意思 我就记得有印象 但又找不到

closed 2025-06-09 by bert1020